ThomasToka/MarlinFirmware #1: Octoprint problem.

Brain7510, 2023-03-18:
Hi!
I uploaded this firmware (firmware-20230315-150030-E3S1Pro-abl5x5-LA-F4.bin) to my Ender 3 S1 Pro.
I have use octoprint and some plugins.
After the firmware upgrade I made a factory reset on the printer, and adjusted the Z offset.
After this I try to connect with octoprint to the printer, but when I try to start run G29 the printer is lost the connection with octoprint.
The connection setting is same as with the original firmware.
[image]
Whet I tried to mesh leveling with bed level plugin, also happened this.

What can I do ?

ThomasToka, 2023-03-18:
Hi,
thanks for the report.
Can you send the terminal output please?
If you do it via the touch it works?
Will try to reproduce and have a look in the code.

ThomasToka, 2023-03-18:
First test:

Abl started with g29 via terminal in pronterface:

`Recv: Bilinear Leveling Grid:
Recv:       0      1      2      3      4
Recv:  0 -0.072 -0.045 -0.095 -0.147 -0.192
Recv:  1 -0.032 +0.015 +0.003 -0.055 -0.132
Recv:  2 +0.023 +0.045 +0.018 -0.040 -0.175
Recv:  3 +0.024 +0.062 +0.038 -0.010 -0.115
Recv:  4 -0.005 +0.040 +0.025 -0.050 -0.178
Recv: `

ThomasToka, 2023-03-18:
Second test:

Abl started in Terminal of octoprint.
Works also.

Third test:
Abl started in Octoprints Bed Visualizer with this Code:

`
M140 S60 ; starting by heating the bed for nominal mesh accuracy
M117 Homing all axes ; send message to printer display
G28      ; home all axes
M420 S0  ; Turning off bed leveling while probing, if firmware is set
         ; to restore after G28
M117 Heating the bed ; send message to printer display
M190 S60 ; waiting until the bed is fully warmed up
M300 S1000 P500 ; chirp to indicate bed mesh levels is initializing
M117 Creating the bed mesh levels ; send message to printer display
M155 S30 ; reduce temperature reporting rate to reduce output pollution
@BEDLEVELVISUALIZER	; tell the plugin to watch for reported mesh
G29 T	   ; run bilinear probing
M155 S3  ; reset temperature reporting
M140 S0 ; cooling down the bed
M500 ; store mesh in EEPROM
M300 S440 P200 ; make calibration completed tones
M300 S660 P250
M300 S880 P300
M117 Bed mesh levels completed ; send message to printer display`

[image]

Brain7510, 2023-03-18:
> Hi, thanks for the report. Can you send the terminal output please? If you do it via the touch it works? Will try to reproduce and have a look in the code.

The bigest problem, is the next.
I put it back the original firmware, and the result is the same.
When I give any command to printer, the printer is start the command and some second later the touch screen is freeze on the menu item where it was.
The octoprint send the same error message.
I don't know what happened, because yesterday evening all worked.
Now I try to install virgin octoprint.

ThomasToka, 2023-03-18:
Sorry for you. But its probably not a firmware problem. The ribbon cable is known to break :-(
I have around 30 testers that checked various versions and as of now all known bugs are fixed. 
Check also the little cr touch cable on the extruder from cr touch to the board. One user had it losen and had something like you. Turn printer off. Just pull it out gently and push in again.

ThomasToka, 2023-03-18:
[image]

Thats the pinout. maybe you can measure your cable with a multimeter.

Brain7510, 2023-03-18:
> Sorry for you. But its probably not a firmware problem. The ribbon cable is known to break :-( I have around 30 testers that checked various versions and as of now all known bugs are fixed. Check [...]

I checked everything. All are good.
Some minutes ago I put back the original firmware and installed a new octoprint and everything are fine.

Now I upgraded the printer with your firmware, and tried to use with virgin octoprint , and I get again that error message.
With original firmware and with the virgin octoprint the printer is working fine.
With your firmware and virgin octoprint I get that error message.

I don't know what is the problem.

ThomasToka, 2023-03-18:
Please sent a output of M503, your octoprint plugin start code and a output of the terminal when the error occurs.

You connect with 115200 ?

ThomasToka, 2023-03-18:
And another thing: What is your screen version shown on the "about" page of your printer?

Brain7510, 2023-03-18:
> Please sent a output of M503, your octoprint plugin start code and a output of the terminal when the error occurs.
> 
> You connect with 115200 ?

Here are lines after G29
The printer is not moved.

Send: G29
Recv: echo:Home XYZ First
Recv: //action:prompt_end
Recv: ok P63 B63
Recv:  T:26.40 /0.00 (3895.69) B:24.77 /0.00 (3909.50) @:0 B@:0
Recv:  T:26.38 /0.00 (3895.88) B:24.83 /0.00 (3909.06) @:0 B@:0
Recv:  T:26.33 /0.00 (3896.31) B:24.72 /0.00 (3909.81) @:0 B@:0
Recv:  T:26.31 /0.00 (3896.44) B:24.68 /0.00 (3910.06) @:0 B@:0
Recv:  T:26.33 /0.00 (3896.31) B:24.67 /0.00 (3910.13) @:0 B@:0
Recv:  T:26.29 /0.00 (3896.69) B:24.60 /0.00 (3910.56) @:0 B@:0
Recv:  T:26.30 /0.00 (3896.56) B:24.61 /0.00 (3910.50) @:0 B@:0
Recv:  T:26.24 /0.00 (3897.13) B:24.67 /0.00 (3910.13) @:0 B@:0
Recv:  T:26.10 /0.00 (3898.31) B:24.60 /0.00 (3910.56) @:0 B@:0
Recv:  T:26.16 /0.00 (3897.75) B:24.58 /0.00 (3910.69) @:0 B@:0
Recv:  T:26.16 /0.00 (3897.81) B:24.61 /0.00 (3910.50) @:0 B@:0
Recv:  T:26.16 /0.00 (3897.75) B:24.72 /0.00 (3909.81) @:0 B@:0
Recv:  T:26.10 /0.00 (3898.31) B:24.63 /0.00 (3910.38) @:0 B@:0
Recv:  T:26.12 /0.00 (3898.19) B:24.68 /0.00 (3910.06) @:0 B@:0
Recv:  T:26.11 /0.00 (3898.25) B:24.64 /0.00 (3910.31) @:0 B@:0
Recv:  T:26.09 /0.00 (3898.44) B:24.68 /0.00 (3910.06) @:0 B@:0
Recv:  T:26.09 /0.00 (3898.44) B:24.59 /0.00 (3910.63) @:0 B@:0
Recv:  T:26.07 /0.00 (3898.63) B:24.60 /0.00 (3910.56) @:0 B@:0
Recv:  T:26.00 /0.00 (3898.75) B:24.59 /0.00 (3910.69) @:0 B@:0
Recv:  T:26.03 /0.00 (3898.94) B:24.64 /0.00 (3910.31) @:0 B@:0
Recv:  T:26.04 /0.00 (3898.63) B:24.59 /0.00 (3910.81) @:0 B@:0
Recv:  T:26.02 /0.00 (3899.00) B:24.62 /0.00 (3910.44) @:0 B@:0
Recv:  T:26.00 /0.00 (3899.19) B:24.64 /0.00 (3910.31) @:0 B@:0
Recv:  T:26.06 /0.00 (3898.69) B:24.54 /0.00 (3910.94) @:0 B@:0
Recv:  T:26.02 /0.00 (3899.00) B:24.56 /0.00 (3910.81) @:0 B@:0

Brain7510, 2023-03-18:
> And another thing: What is your screen version shown on the "about" page of your printer?

1.0.3

ThomasToka, 2023-03-18:
> Send: G29
> Recv: echo:Home XYZ First

sent g28 first.

ThomasToka, 2023-03-18:
> > And another thing: What is your screen version shown on the "about" page of your printer?
> 
> 1.0.3

Thats ok. So this matches.

Brain7510, 2023-03-18:
> > > And another thing: What is your screen version shown on the "about" page of your printer?
> > 
> > 
> > 1.0.3
> 
> Thats ok. So this matches.

When I send G28 the printer is switching on the hotend cooler.

Brain7510, 2023-03-18:
[image]

ThomasToka, 2023-03-18:
turn it of, turn it on. reconnect. sent G28.

An please show output of m503.

Brain7510, 2023-03-18:
> turn it off, turn it on. reconnect. sent G28.
> 
> And please show output of m503.

I sent M503, here is the result:
Send: M503
Recv: echo:; Linear Units:
Recv: echo:  G21 ; (mm)
Recv: echo:; Temperature Units:
Recv: echo:  M149 C ; Units in Celsius
Recv: echo:; Filament settings (Disabled):
Recv: echo:  M200 S0 D1.75
Recv: echo:; Steps per unit:
Recv: echo:  M92 X80.00 Y80.00 Z400.00 E424.90
Recv: echo:; Max feedrates (units/s):
Recv: echo:  M203 X1000.00 Y1000.00 Z10.00 E50.00
Recv: echo:; Max Acceleration (units/s2):
Recv: echo:  M201 X1000.00 Y1000.00 Z100.00 E1000.00
Recv: echo:; Acceleration (units/s2) (P<print-accel> R<retract-accel> T<travel-accel>):
Recv: echo:  M204 P1000.00 R1000.00 T1500.00
Recv: echo:; Advanced (B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_jerk> Y<max_jerk> Z<max_jerk> E<max_jerk>):
Recv: echo:  M205 B50000.00 S0.00 T0.00 X8.00 Y8.00 Z0.40 E5.00
Recv: echo:; Home offset:
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:; Auto Bed Leveling:
Recv: echo:  M420 S0 Z5.00 ; Leveling OFF
Recv: echo:; Material heatup parameters:
Recv: echo:  M145 S0 H210.00 B60.00 F255
Recv: echo:  M145 S1 H240.00 B100.00 F255
Recv: echo:  M145 S2 H230.00 B80.00 F128
Recv: echo:  M145 S3 H190.00 B50.00 F128
Recv: echo:; Hotend PID:
Recv: echo:  M301 P20.13 I1.86 D54.35
Recv: echo:; Bed PID:
Recv: echo:  M304 P127.12 I23.03 D476.81
Recv: echo:; Power-loss recovery:
Recv: echo:  M413 S0 ; OFF
Recv: echo:; Z-Probe Offset:
Recv: echo:  M851 Probe Offset X-31.50 Y-41.80 Z0.00 ; (mm)
Recv: echo:; Input Shaping:
Recv: echo:  M593 X F40.00 D0.15
Recv: echo:  M593 Y F50.00 D0.15
Recv: echo:; Linear Advance:
Recv: echo:  M900 K0.05
Recv: echo:; Filament load/unload:
Recv: echo:  M603 L0.00 U100.00 ; (mm)
Recv: echo:; Filament runout sensor:
Recv: echo:  M412 S1 ; Sensor ON
Recv: ok P63 B63
Recv:  T:23.61 /0.00 (3916.88) B:23.17 /0.00 (3919.69) @:0 B@:0
Recv:  T:23.76 /0.00 (3915.94) B:23.12 /0.00 (3920.06) @:0 B@:0
Recv:  T:23.80 /0.00 (3915.69) B:23.14 /0.00 (3919.88) @:0 B@:0
Recv:  T:23.69 /0.00 (3916.38) B:23.14 /0.00 (3919.88) @:0 B@:0
Recv:  T:23.72 /0.00 (3916.19) B:23.19 /0.00 (3919.56) @:0 B@:0

Brain7510, 2023-03-18:
I sent G28.
Here is the result:
Send: G28
Recv: echo:busy: processing
Printer seems to support the busy protocol, will adjust timeouts and set busy interval accordingly
Recv:  T:23.46 /0.00 (3917.88) B:23.05 /0.00 (3920.50) @:0 B@:0
Recv:  T:23.47 /0.00 (3917.81) B:22.91 /0.00 (3921.38) @:0 B@:0
Recv: echo:busy: processing
Recv:  T:23.52 /0.00 (3917.50) B:22.89 /0.00 (3921.50) @:0 B@:0
Recv: echo:busy: processing
Recv:  T:23.42 /0.00 (3918.13) B:22.91 /0.00 (3921.38) @:0 B@:0
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv:  T:23.50 /0.00 (3917.63) B:23.05 /0.00 (3920.50) @:0 B@:0
Recv: echo:busy: processing
Recv:  T:23.39 /0.00 (3918.31) B:22.91 /0.00 (3921.38) @:0 B@:0
Recv: echo:busy: processing
Recv:  T:23.41 /0.00 (3918.19) B:22.97 /0.00 (3921.00) @:0 B@:0
Recv:  T:23.83 /0.00 (3915.50) B:23.26 /0.00 (3919.13) @:0 B@:0
Recv: echo:busy: processing
Recv: X:148.50 Y:158.80 Z:5.00 E:0.00 Count X:11880 Y:12704 Z:2000
Recv: echo:Invalid mesh.
Recv: Error:Failed to enable Bed Leveling
Changing monitoring state from "Operational" to "Error"
Send: M112
Send: N2 M112*35
Send: N3 M104 T0 S0*34
Send: N4 M140 S0*97
Changing monitoring state from "Error" to "Offline after error"
Connection closed, closing down monitor

ThomasToka, 2023-03-18:
Ok thats weird.

Please do the following:

Disconnect octoprint, restart printer, go to leveling, start autoleveling.

ThomasToka, 2023-03-18:
`Recv: echo:Fade Height 5.00
Recv: echo:Invalid mesh.
Recv: Error:Failed to enable Bed Leveling
WARNING! Received an error from the printer's firmware, ignoring that as configured but you might want to investigate what happened here! Error: Failed to enable Bed Leveling
Recv: echo:Bed Leveling OFF
Recv: echo:Fade Height 5.00`

ok the error is ok and normal if there is no mesh.

so ignore it. 

you printer homed when you made g28 ?

Then just process with g29.

ThomasToka, 2023-03-18:
But that your printer goes offline not.

Brain7510, 2023-03-18:
> Ok thats weird.
> 
> Please do the following:
> 
> Disconnect octoprint, restart printer, go to leveling, start autoleveling.

It is done.

ThomasToka, 2023-03-18:
ok. thats good. 

then reconnect with octoprint an show m503 again, it should contain a mesh now.

Brain7510, 2023-03-18:
Thansk dude !

I get a message for octoprint.
G29 Retrying.
It is normal ?

ThomasToka, 2023-03-18:
Dont know what happens with your machine. Octoprint is a gcode sender in this case.. So i cant say what it sends as i dont see it.

Brain7510, 2023-03-18:
Send: M503
Recv: echo:; Linear Units:
Recv: echo:  G21 ; (mm)
Recv: echo:; Temperature Units:
Recv: echo:  M149 C ; Units in Celsius
Recv: echo:; Filament settings (Disabled):
Recv: echo:  M200 S0 D1.75
Recv: echo:; Steps per unit:
Recv: echo:  M92 X80.00 Y80.00 Z400.00 E424.90
Recv: echo:; Max feedrates (units/s):
Recv: echo:  M203 X1000.00 Y1000.00 Z10.00 E50.00
Recv: echo:; Max Acceleration (units/s2):
Recv: echo:  M201 X1000.00 Y1000.00 Z100.00 E1000.00
Recv: echo:; Acceleration (units/s2) (P<print-accel> R<retract-accel> T<travel-accel>):
Recv: echo:  M204 P1000.00 R1000.00 T1500.00
Recv: echo:; Advanced (B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_jerk> Y<max_jerk> Z<max_jerk> E<max_jerk>):
Recv: echo:  M205 B50000.00 S0.00 T0.00 X8.00 Y8.00 Z0.40 E5.00
Recv: echo:; Home offset:
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:; Auto Bed Leveling:
Recv: echo:  M420 S1 Z5.00 ; Leveling ON
Recv: echo:  G29 W I0 J0 Z-0.22750
Recv: echo:  G29 W I1 J0 Z-0.17500
Recv: echo:  G29 W I2 J0 Z-0.15250
Recv: echo:  G29 W I3 J0 Z-0.20500
Recv: echo:  G29 W I4 J0 Z-0.28500
Recv: echo:  G29 W I0 J1 Z-0.09750
Recv: echo:  G29 W I1 J1 Z-0.04750
Recv: echo:  G29 W I2 J1 Z-0.01500
Recv: echo:  G29 W I3 J1 Z-0.05500
Recv: echo:  G29 W I4 J1 Z-0.11750
Recv: echo:  G29 W I0 J2 Z-0.07500
Recv: echo:  G29 W I1 J2 Z-0.02500
Recv: echo:  G29 W I2 J2 Z0.00000
Recv: echo:  G29 W I3 J2 Z-0.03000
Recv: echo:  G29 W I4 J2 Z-0.08000
Recv: echo:  G29 W I0 J3 Z-0.03500
Recv: echo:  G29 W I1 J3 Z0.00750
Recv: echo:  G29 W I2 J3 Z0.03250
Recv: echo:  G29 W I3 J3 Z-0.00250
Recv: echo:  G29 W I4 J3 Z-0.05500
Recv: echo:  G29 W I0 J4 Z-0.09875
Recv: echo:  G29 W I1 J4 Z-0.04750
Recv: echo:  G29 W I2 J4 Z-0.01000
Recv: echo:  G29 W I3 J4 Z-0.03750
Recv: echo:  G29 W I4 J4 Z-0.08250
Recv: echo:; Material heatup parameters:
Recv: echo:  M145 S0 H210.00 B60.00 F255
Recv: echo:  M145 S1 H240.00 B100.00 F255
Recv: echo:  M145 S2 H230.00 B80.00 F128
Recv: echo:  M145 S3 H190.00 B50.00 F128
Recv: echo:; Hotend PID:
Recv: echo:  M301 P20.13 I1.86 D54.35
Recv: echo:; Bed PID:
Recv: echo:  M304 P127.12 I23.03 D476.81
Recv: echo:; Power-loss recovery:
Recv: echo:  M413 S0 ; OFF
Recv: echo:; Z-Probe Offset:
Recv: echo:  M851 Probe Offset X-31.50 Y-41.80 Z0.00 ; (mm)
Recv: echo:; Input Shaping:
Recv: echo:  M593 X F40.00 D0.15
Recv: echo:  M593 Y F50.00 D0.15
Recv: echo:; Linear Advance:
Recv: echo:  M900 K0.05
Recv: echo:; Filament load/unload:
Recv: echo:  M603 L0.00 U100.00 ; (mm)
Recv: echo:; Filament runout sensor:
Recv: echo:  M412 S1 ; Sensor ON
Recv: ok P63 B63
Recv:  T:23.51 /0.00 (3917.56) B:21.92 /0.00 (3927.69) @:0 B@:0
Recv:  T:23.59 /0.00 (3917.00) B:21.97 /0.00 (3927.38) @:0 B@:0
Recv:  T:23.53 /0.00 (3917.44) B:21.95 /0.00 (3927.50) @:0 B@:0
Recv:  T:23.55 /0.00 (3917.25) B:22.03 /0.00 (3927.00) @:0 B@:0
Recv:  T:23.58 /0.00 (3917.06) B:21.94 /0.00 (3927.56) @:0 B@:0
Recv:  T:23.54 /0.00 (3917.38) B:21.94 /0.00 (3927.56) @:0 B@:0
Recv:  T:23.55 /0.00 (3917.25) B:21.95 /0.00 (3927.50) @:0 B@:0
Recv:  T:23.59 /0.00 (3917.00) B:22.03 /0.00 (3927.00) @:0 B@:0

ThomasToka, 2023-03-18:
Ok so it works. I really dont know whats going on. I can only say that a broken cable makes weird things. And that it works with stock firmware "without this" i cant say why. the cable is in movement and not always looses the broken connection. If it really should be hardware then you will also have it with stock firmware soon. it just depends on the movements made.

A bug in the firmware is not visible for me atm :(

ThomasToka, 2023-03-18:
And on thing: If you use octoprint, then use it only and dont for example try to stop the print on the screen. octoprint sends further commands and you printer will go nuts ^^

Brain7510, 2023-03-18:
When I try to print any sample, the printer head go lover than the heated bed, and 5 mm left than the heated bed.
What I must set to print to the heated bed, and don't go under the bed ?

ThomasToka, 2023-03-18:
You have not readen the instructions.

Check your slicer start code. Your bed starts now at x=0 as it should be.
So if your have purge lines with x=-5 it wont work.

ThomasToka, 2023-03-18:
`
G1 X5.1 Y20 Z0.28 F5000.0 ;Move to start position
G1 X5.1 Y200.0 Z0.28 F1500.0 E15 ;Draw the first line
G1 X5.4 Y200.0 Z0.28 F5000.0 ;Move to side a little
G1 X5.4 Y20 Z0.28 F1500.0 E30 ;Draw the second line
`

like this..

Brain7510, 2023-03-18:
With the original firmware the zoffset was -3,6mm
Now with your firmware the zoffset is -4.07mm

Why ?

Brain7510, 2023-03-18:
Here is my gcode rfom the slicer

M117 Getting the bed up to temp!
M140 S{material_bed_temperature_layer_0} ; Set Heat Bed temperature
M190 S{material_bed_temperature_layer_0} ; Wait for Heat Bed temperature
M117 Pre-heating the extruder!
M104 S190; start warming extruder to 190
G28 ; Home all axes
G1 Z10
G1 Y0
G1 X0
; M117 Auto bed-level GO!
; G29 ; Auto bed-level (BL-Touch)
M420 S1 ; Load bed level mesh
G92 E0 ; Reset Extruder
; M117 Getting the extruder up to temp!
M104 S{material_print_temperature_layer_0} ; Set Extruder temperature
M109 S{material_print_temperature_layer_0} ; Wait for Extruder temperature
G1 Z1.0 F3000 ; move z up little to prevent scratching of surface
G1 X0.1 Y20 Z0.3 F5000.0 ; move to start-line position
M300 P250 ; play chime to indicate print starting
G1 X0.1 Y200.0 Z0.3 F1500.0 E15 ; draw 1st line
G1 X0.4 Y200.0 Z0.3 F5000.0 ; move to side a little
G1 X0.4 Y20 Z0.3 F1500.0 E30 ; draw 2nd line
G92 E0 ; reset extruder
G1 Z1.0 F3000 ; move z up little to prevent scratching of surface
; End of custom start GCode

ThomasToka, 2023-03-18:
`
G1 X5.1 Y20 Z0.3 F5000.0 ; move to start-line position
M300 P250 ; play chime to indicate print starting
G1 X5.1 Y200.0 Z0.3 F1500.0 E15 ; draw 1st line
G1 X5.4 Y200.0 Z0.3 F5000.0 ; move to side a little
G1 X5.4 Y20 Z0.3 F1500.0 E30 ; draw 2nd line
`

as for the offset. maybe. a few testers had other values. double check when your printer starts printing.

And please do the following:

G1 X0 F5000

Is your nozzle then right over the bed edge?

ThomasToka, 2023-03-18:
And remove this:

G1 Z10

G1 Y0

G1 X0

ThomasToka, 2023-03-18:
Mine is as simple as that:

`
G28 ;Home

G92 E0 ;Reset Extruder

M420 S1 Z10; Abl Not needed but does not hurt

G1 Z2.0 F3000 ;Move Z Axis up

G1 X5.1 Y20 Z0.28 F5000.0 ;Move to start position

G1 X5.1 Y200.0 Z0.28 F1500.0 E15 ;Draw the first line

G1 X5.4 Y200.0 Z0.28 F5000.0 ;Move to side a little

G1 X5.4 Y20 Z0.28 F1500.0 E30 ;Draw the second line

G92 E0 ;Reset Extruder

G1 Z2.0 F3000 ;Move Z Axis up

`

Brain7510, 2023-03-18:
> And please do the following:
> 
> G1 X0 F5000

"And please do the following:

G1 X0 F5000"

Sorry i don't understand this.

I will try the new gcode, but now we must leave home.

Brain7510, 2023-03-18:
Here is the repaired Gcode from the slicer:

M117 Getting the bed up to temp!
M140 S{material_bed_temperature_layer_0} ; Set Heat Bed temperature
M190 S{material_bed_temperature_layer_0} ; Wait for Heat Bed temperature
M117 Pre-heating the extruder!
M104 S190; start warming extruder to 190
G28 ; Home all axes
 
; M117 Auto bed-level GO!
; G29 ; Auto bed-level (BL-Touch)
M420 S1 ; Load bed level mesh
G92 E0 ; Reset Extruder
; M117 Getting the extruder up to temp!
M104 S{material_print_temperature_layer_0} ; Set Extruder temperature
M109 S{material_print_temperature_layer_0} ; Wait for Extruder temperature
G1 Z2.0 F3000 ; move z up little to prevent scratching of surface
G1 X5.1 Y20 Z0.28 F5000.0 ;Move to start position
M300 P250 ; play chime to indicate print starting
G1 X5.1 Y200.0 Z0.28 F1500.0 E15 ;Draw the first line
G1 X5.4 Y200.0 Z0.28 F5000.0 ;Move to side a little
G1 X5.4 Y20 Z0.28 F1500.0 E30 ;Draw the second line
G92 E0 ; reset extruder
G1 Z2.0 F3000 ; move z up little to prevent scratching of surface
; End of custom start GCode

ThomasToka, 2023-03-18:
yes should work like this. i prefer to set a leveling height of 10mm.. like this 

M420 S1 Z10; Load bed level mesh

Brain7510, 2023-03-19:
"G1 X0 F5000
Is your nozzle then right over the bed edge?"

Yes

ThomasToka, 2023-03-19:
the you should be fine.

Brain7510, 2023-03-19:
Now I setting up everything again. The printing is fine, but he Z axis motor is very hot. Hoter than with original firmware. It is normal ?
Thanks

ThomasToka, 2023-03-19:
yes thats known and simple as the fact that the accelerations are a bit higher with my build.

you could if you are able to measure your stepper motors vref. creality usually has is set a bit to high.

but this little more temperature wont hurt your motors. the can stand way more ;)

Brain7510, 2023-03-19:
One other question. 
On the touchscreen the duration % is not changing under print if I start the printing from octoprint.
Do you have a plan for repairing this ? Is it possible ?

ThomasToka, 2023-03-19:
Indeed you are right here. This was fixed already, but the fix got lost on my last rebase of the code.

Sorry for this and thank you for finding and reporting.

Will upload new bins in some minutes.

ThomasToka, 2023-03-19:
Could you please verify with this file: https://github.com/ThomasToka/MarlinFirmware/raw/Firmware-Binaries/firmware-20230319-182611-E3S1Pro-abl5x5-LA-F4.bin

Brain7510, 2023-03-19:
> Could you please verify with this file: https://github.com/ThomasToka/MarlinFirmware/raw/Firmware-Binaries/firmware-20230319-182611-E3S1Pro-abl5x5-LA-F4.bin

What can I do after update ? I must setup again the z-offset and I must do again the extruder calibration ?

ThomasToka, 2023-03-19:
you can note you offset, update firmware, and sent M871 z-1.11 and M500, to save it -1.11 is just a example..

heat up, autoleveling, start print.

Brain7510, 2023-03-19:
> you can note you offset, update firmware, and sent M871 z-1.11 and M500, to save it -1.11 is just a example..
> 
> heat up, autoleveling, start print.

Sorry but    M871 - Probe temperature config.
M851 is the correct I think.

ThomasToka, 2023-03-19:
ups yes sorry.

ThomasToka, 2023-03-19:
all updates pushed already.

Brain7510, 2023-03-19:
I updated the firmware.
Some reporting:
1. Before the update the Z-offset was 3.00 mm. After the update I changed it to 3.24 because with 3.24 was good.
2. All of the settings missed from ADV.SET. / movement . I have done a factory reset and the datas came back.
[image]

The duration % is working as you see on picture.

One question: Is it possible to move the duration % to the original circle on the center of the display ?

Thanks your work.

ThomasToka, 2023-03-19:
Yes you need to install the "M73 progress" octoprint plugin. Octoprint then sends the data.

ThomasToka, 2023-03-19:
But you will see differences what is shown on the top and what the wheel shows and what percentage is show in the uppper line. Which is unfixable atm.

ThomasToka, 2023-03-19:
And to your offset: Other testers had the springs not tight enough so the bed had different offset in each messurement.

After tightening them it went away. I always have the same offset if i install new builds and the buildplate and wheels were not touched.

closed 2023-03-20 by ThomasToka