ThomasToka/MarlinFirmware #2: Freezes with invalid mesh

andre68723, 2023-03-29:
Hi,

looks similar to #1 

Followed your instructions, noted down my -2.70 offset.
Display also not reacting anymore 

Firmware is `firmware-20230326-220058-E3S1Pro-abl5x5-LA-F4.bin`

Thanks, Andre

```
Recv:  T:120.02 /0.00 (1779.50) B:51.85 /0.00 (3579.50) @:0 B@:0
Recv:  T:119.21 /0.00 (1802.25) B:51.76 /0.00 (3580.94) @:0 B@:0
Recv:  T:118.35 /0.00 (1826.25) B:51.66 /0.00 (3582.75) @:0 B@:0
Send: M502
Recv: //action:notification Hardcoded Default Settings Loaded
Recv: echo:Hardcoded Default Settings Loaded
Recv: ok P63 B63
Recv:  T:117.55 /0.00 (1848.50) B:51.60 /0.00 (3583.81) @:0 B@:0
Recv:  T:116.69 /0.00 (1872.81) B:51.44 /0.00 (3586.69) @:0 B@:0
Recv:  T:115.91 /0.00 (1894.63) B:51.40 /0.00 (3587.44) @:0 B@:0
Recv:  T:115.04 /0.00 (1919.00) B:51.25 /0.00 (3590.00) @:0 B@:0
Send: M851 Z-2.70
Recv: ok P63 B63
Recv:  T:114.25 /0.00 (1941.56) B:51.20 /0.00 (3590.81) @:0 B@:0
Recv:  T:113.44 /0.00 (1964.88) B:51.16 /0.00 (3591.63) @:0 B@:0
Recv:  T:112.68 /0.00 (1986.81) B:51.01 /0.00 (3594.25) @:0 B@:0
Recv:  T:111.87 /0.00 (2010.06) B:50.96 /0.00 (3595.06) @:0 B@:0
Send: M500
Recv: echo:Settings Stored (712 bytes; crc 57696)
Recv: //action:notification Settings Stored
Recv: ok P63 B63
Recv:  T:111.13 /0.00 (2031.38) B:50.85 /0.00 (3597.06) @:0 B@:0
Recv:  T:110.28 /0.00 (2055.81) B:50.77 /0.00 (3598.38) @:0 B@:0
Recv:  T:109.54 /0.00 (2077.75) B:50.70 /0.00 (3599.63) @:0 B@:0
Send: M503
Recv: echo:; Linear Units:
Recv: echo:  G21 ; (mm)
Recv: echo:; Temperature Units:
Recv: echo:  M149 C ; Units in Celsius
Recv: echo:; Filament settings (Disabled):
Recv: echo:  M200 S0 D1.75
Recv: echo:; Steps per unit:
Recv: echo:  M92 X80.00 Y80.00 Z400.00 E424.90
Recv: echo:; Max feedrates (units/s):
Recv: echo:  M203 X1000.00 Y1000.00 Z10.00 E50.00
Recv: echo:; Max Acceleration (units/s2):
Recv: echo:  M201 X1000.00 Y1000.00 Z100.00 E1000.00
Recv: echo:; Acceleration (units/s2) (P<print-accel> R<retract-accel> T<travel-accel>):
Recv: echo:  M204 P1000.00 R1000.00 T1500.00
Recv: echo:; Advanced (B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_jerk> Y<max_jerk> Z<max_jerk> E<max_jerk>):
Recv: echo:  M205 B50000.00 S0.00 T0.00 X8.00 Y8.00 Z0.40 E5.00
Recv: echo:; Home offset:
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:; Auto Bed Leveling:
Recv: echo:  M420 S0 Z5.00 ; Leveling OFF
Recv: echo:; Material heatup parameters:
Recv: echo:  M145 S0 H210.00 B60.00 F255
Recv: echo:  M145 S1 H240.00 B100.00 F255
Recv: echo:  M145 S2 H230.00 B80.00 F128
Recv: echo:  M145 S3 H190.00 B50.00 F128
Recv: echo:; Hotend PID:
Recv: echo:  M301 P20.13 I1.86 D54.35
Recv: echo:; Bed PID:
Recv: echo:  M304 P127.12 I23.03 D476.81
Recv: echo:; Power-loss recovery:
Recv: echo:  M413 S0 ; OFF
Recv: echo:; Z-Probe Offset:
Recv: echo:  M851 Probe Offset X-31.50 Y-41.80 Z-2.70 ; (mm)
Recv: echo:; Input Shaping:
Recv: echo:  M593 X F40.00 D0.15
Recv: echo:  M593 Y F50.00 D0.15
Recv: echo:; Linear Advance:
Recv: echo:  M900 K0.05
Recv: echo:; Filament load/unload:
Recv: echo:  M603 L0.00 U100.00 ; (mm)
Recv: echo:; Filament runout sensor:
Recv: echo:  M412 S1 ; Sensor ON
Recv: ok P63 B63
Recv:  T:108.76 /0.00 (2100.75) B:50.61 /0.00 (3601.19) @:0 B@:0
Recv:  T:108.02 /0.00 (2122.63) B:50.53 /0.00 (3602.63) @:0 B@:0
Recv:  T:107.24 /0.00 (2145.63) B:50.41 /0.00 (3604.75) @:0 B@:0
Recv:  T:106.54 /0.00 (2166.44) B:50.37 /0.00 (3605.56) @:0 B@:0
Recv:  T:105.76 /0.00 (2189.44) B:50.28 /0.00 (3607.00) @:0 B@:0
Recv:  T:105.07 /0.00 (2209.88) B:50.25 /0.00 (3607.69) @:0 B@:0
Recv:  T:104.36 /0.00 (2231.44) B:50.11 /0.00 (3610.00) @:0 B@:0
Send: G28
Recv: echo:busy: processing
Printer seems to support the busy protocol, will adjust timeouts and set busy interval accordingly
Recv:  T:103.46 /0.00 (2258.69) B:50.02 /0.00 (3611.63) @:0 B@:0
Recv: echo:busy: processing
Recv:  T:102.75 /0.00 (2280.44) B:49.91 /0.00 (3613.38) @:0 B@:0
Recv:  T:102.11 /0.00 (2300.00) B:49.81 /0.00 (3614.94) @:0 B@:0
Recv: echo:busy: processing
Recv:  T:101.39 /0.00 (2321.69) B:49.70 /0.00 (3616.56) @:0 B@:0
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv:  T:100.66 /0.00 (2344.00) B:49.65 /0.00 (3617.25) @:0 B@:0
Recv:  T:99.98 /0.00 (2364.50) B:49.56 /0.00 (3618.75) @:0 B@:0
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv:  T:99.32 /0.00 (2384.00) B:49.52 /0.00 (3619.25) @:0 B@:0
Recv: echo:busy: processing
Recv:  T:98.62 /0.00 (2404.81) B:49.34 /0.00 (3622.06) @:0 B@:0
Recv: X:148.50 Y:158.80 Z:5.00 E:0.00 Count X:11880 Y:12704 Z:2000
Recv: echo:Invalid mesh.
Recv: Error:Failed to enable Bed Leveling
Changing monitoring state from "Operational" to "Error"
Send: M112
Send: N2 M112*35
Send: N3 M104 T0 S0*34
Send: N4 M140 S0*97
Changing monitoring state from "Error" to "Offline after error"
Connection closed, closing down monitor
```

[image]

ThomasToka, 2023-03-29:
hi, will have a look into it in the next 24h. 
thanks for reporting..

andre68723, 2023-03-29:
yes, reproducible.

retried from powercycle -> connect to G28

```
Changing monitoring state from "Offline" to "Opening serial connection"
Connecting to port /dev/ttyUSB0, baudrate 115200
Changing monitoring state from "Opening serial connection" to "Connecting"
Connected to: Serial<id=0x7f78047490, open=True>(port='/dev/ttyUSB0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Send: N0 M110 N0*125
Recv: ok N0 P63 B63
Send: N0 M110 N0*125
Changing monitoring state from "Connecting" to "Operational"
Recv: ok N0 P63 B63
Send: N0 M110 N0*125
Recv: ok N0 P63 B63
Send: N1 M115*39
Recv: FIRMWARE_NAME:Marlin 2.1.x-F4-byTT (Mar 26 2023 22:01:29) SOURCE_CODE_URL:github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:Ender-3 S1 Pro EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
Recv: Cap:SERIAL_XON_XOFF:0
Recv: Cap:BINARY_FILE_TRANSFER:0
Recv: Cap:EEPROM:1
Recv: Cap:VOLUMETRIC:1
Recv: Cap:AUTOREPORT_POS:0
Recv: Cap:AUTOREPORT_TEMP:1
Recv: Cap:PROGRESS:0
Recv: Cap:PRINT_JOB:1
Recv: Cap:AUTOLEVEL:1
Recv: Cap:RUNOUT:1
Recv: Cap:Z_PROBE:1
Recv: Cap:LEVELING_DATA:1
Recv: Cap:BUILD_PERCENT:1
Recv: Cap:SOFTWARE_POWER:0
Recv: Cap:TOGGLE_LIGHTS:0
Recv: Cap:CASE_LIGHT_BRIGHTNESS:0
Recv: Cap:LASER:1
Recv: Cap:EMERGENCY_PARSER:1
Recv: Cap:HOST_ACTION_COMMANDS:1
Recv: Cap:PROMPT_SUPPORT:1
Recv: Cap:SDCARD:1
Recv: Cap:MULTI_VOLUME:0
Recv: Cap:REPEAT:0
Recv: Cap:SD_WRITE:1
Recv: Cap:AUTOREPORT_SD_STATUS:0
Recv: Cap:LONG_FILENAME:1
Recv: Cap:LFN_WRITE:0
Recv: Cap:CUSTOM_FIRMWARE_UPLOAD:0
Recv: Cap:EXTENDED_M20:1
Recv: Cap:THERMAL_PROTECTION:1
Recv: Cap:MOTION_MODES:0
Recv: Cap:ARCS:1
Recv: Cap:BABYSTEPPING:1
Recv: Cap:CHAMBER_TEMPERATURE:0
Recv: Cap:COOLER_TEMPERATURE:0
Recv: Cap:MEATPACK:0
Recv: Cap:CONFIG_EXPORT:0
Recv: area:{full:{min:{x:-0.80,y:-2.00,z:0.00},max:{x:235.00,y:235.00,z:250.00}},work:{min:{x:0.00,y:0.00,z:0.00},max:{x:235.00,y:235.00,z:250.00}}}
Recv: ok N1 P63 B63
Send: M21
Recv: echo:No SD card
Recv: //action:notification Media Init Fail
Recv: ok P63 B63
Send: M117 Octoprint is connected
Recv: ok P63 B63
Send: M155 S2
Recv: ok P63 B63
Send: M876 P1
Recv: ok P63 B63
[...]
Send: M502
Recv: //action:notification Hardcoded Default Settings Loaded
Recv: echo:Hardcoded Default Settings Loaded
Recv: ok P63 B63
[...]
Send: M851 Z-2.70
Recv: ok P63 B63
[...]
Send: M500
Recv: echo:Settings Stored (712 bytes; crc 57696)
Recv: //action:notification Settings Stored
Recv: ok P63 B63
[...]
Send: M503
Recv: echo:; Linear Units:
Recv: echo:  G21 ; (mm)
Recv: echo:; Temperature Units:
Recv: echo:  M149 C ; Units in Celsius
Recv: echo:; Filament settings (Disabled):
Recv: echo:  M200 S0 D1.75
Recv: echo:; Steps per unit:
Recv: echo:  M92 X80.00 Y80.00 Z400.00 E424.90
Recv: echo:; Max feedrates (units/s):
Recv: echo:  M203 X1000.00 Y1000.00 Z10.00 E50.00
Recv: echo:; Max Acceleration (units/s2):
Recv: echo:  M201 X1000.00 Y1000.00 Z100.00 E1000.00
Recv: echo:; Acceleration (units/s2) (P<print-accel> R<retract-accel> T<travel-accel>):
Recv: echo:  M204 P1000.00 R1000.00 T1500.00
Recv: echo:; Advanced (B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_jerk> Y<max_jerk> Z<max_jerk> E<max_jerk>):
Recv: echo:  M205 B50000.00 S0.00 T0.00 X8.00 Y8.00 Z0.40 E5.00
Recv: echo:; Home offset:
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:; Auto Bed Leveling:
Recv: echo:  M420 S0 Z5.00 ; Leveling OFF
Recv: echo:; Material heatup parameters:
Recv: echo:  M145 S0 H210.00 B60.00 F255
Recv: echo:  M145 S1 H240.00 B100.00 F255
Recv: echo:  M145 S2 H230.00 B80.00 F128
Recv: echo:  M145 S3 H190.00 B50.00 F128
Recv: echo:; Hotend PID:
Recv: echo:  M301 P20.13 I1.86 D54.35
Recv: echo:; Bed PID:
Recv: echo:  M304 P127.12 I23.03 D476.81
Recv: echo:; Power-loss recovery:
Recv: echo:  M413 S0 ; OFF
Recv: echo:; Z-Probe Offset:
Recv: echo:  M851 Probe Offset X-31.50 Y-41.80 Z-2.70 ; (mm)
Recv: echo:; Input Shaping:
Recv: echo:  M593 X F40.00 D0.15
Recv: echo:  M593 Y F50.00 D0.15
Recv: echo:; Linear Advance:
Recv: echo:  M900 K0.05
Recv: echo:; Filament load/unload:
Recv: echo:  M603 L0.00 U100.00 ; (mm)
Recv: echo:; Filament runout sensor:
Recv: echo:  M412 S1 ; Sensor ON
Recv: ok P63 B63
[...]
Send: G28
[...]
Printer seems to support the busy protocol, will adjust timeouts and set busy interval accordingly
[...]
Recv: echo:Invalid mesh.
Recv: Error:Failed to enable Bed Leveling
Changing monitoring state from "Operational" to "Error"
Send: M112
Send: N2 M112*35
Send: N3 M104 T0 S0*34
Send: N4 M140 S0*97
Changing monitoring state from "Error" to "Offline after error"
Connection closed, closing down monitor
```

ThomasToka, 2023-03-29:
can you please test this build https://github.com/ThomasToka/MarlinFirmware/commit/a436de7d9fcc6d641e4e8b2f182bad398330fa99

and sent the same output like you sent already? thank you!

ThomasToka, 2023-03-29:
ok strange. and this https://github.com/ThomasToka/MarlinFirmware/commit/97d15a7676880dcb3b2ba39bbf55780f639ff71b

one more thing: can you factory reset over the touch first before g28 as a try. is there something different?

andre68723, 2023-03-29:
same for reset via touch and same for without m500 -> freezes

if turned back on, Z-2.70 is stored (display) 
and I can execute ABL via touch.

Result:
[image]

doing a G28 afterwards - it's still alive

andre68723, 2023-03-29:
you nailed it

```
Send: M502
Recv: //action:notification Hardcoded Default Settings Loaded
Recv: echo:Hardcoded Default Settings Loaded
Recv: ok P63 B63
[...]
Send: M851 Z-2.70
Recv: ok P63 B63
[...]
Send: M500
Recv: echo:Settings Stored (712 bytes; crc 57696)
Recv: //action:notification Settings Stored
Recv: //action:notification Settings Stored
Recv: ok P63 B63
[...]
Send: M503
Recv: echo:; Linear Units:
Recv: echo:  G21 ; (mm)
Recv: echo:; Temperature Units:
Recv: echo:  M149 C ; Units in Celsius
Recv: echo:; Filament settings (Disabled):
Recv: echo:  M200 S0 D1.75
Recv: echo:; Steps per unit:
Recv: echo:  M92 X80.00 Y80.00 Z400.00 E424.90
Recv: echo:; Max feedrates (units/s):
Recv: echo:  M203 X1000.00 Y1000.00 Z10.00 E50.00
Recv: echo:; Max Acceleration (units/s2):
Recv: echo:  M201 X1000.00 Y1000.00 Z100.00 E1000.00
Recv: echo:; Acceleration (units/s2) (P<print-accel> R<retract-accel> T<travel-accel>):
Recv: echo:  M204 P1000.00 R1000.00 T1500.00
Recv: echo:; Advanced (B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_jerk> Y<max_jerk> Z<max_jerk> E<max_jerk>):
Recv: echo:  M205 B50000.00 S0.00 T0.00 X8.00 Y8.00 Z0.40 E5.00
Recv: echo:; Home offset:
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:; Auto Bed Leveling:
Recv: echo:  M420 S0 Z5.00 ; Leveling OFF
Recv: echo:; Material heatup parameters:
Recv: echo:  M145 S0 H210.00 B60.00 F255
Recv: echo:  M145 S1 H240.00 B100.00 F255
Recv: echo:  M145 S2 H230.00 B80.00 F128
Recv: echo:  M145 S3 H190.00 B50.00 F128
Recv: echo:; Hotend PID:
Recv: echo:  M301 P20.13 I1.86 D54.35
Recv: echo:; Bed PID:
Recv: echo:  M304 P127.12 I23.03 D476.81
Recv: echo:; Power-loss recovery:
Recv: echo:  M413 S0 ; OFF
Recv: echo:; Z-Probe Offset:
Recv: echo:  M851 Probe Offset X-31.50 Y-41.80 Z-2.70 ; (mm)
Recv: echo:; Input Shaping:
Recv: echo:  M593 X F40.00 D0.15
Recv: echo:  M593 Y F50.00 D0.15
Recv: echo:; Linear Advance:
Recv: echo:  M900 K0.05
Recv: echo:; Filament load/unload:
Recv: echo:  M603 L0.00 U100.00 ; (mm)
Recv: echo:; Filament runout sensor:
Recv: echo:  M412 S1 ; Sensor ON
Recv: ok P63 B63
[...]
Send: G28
[...]
Printer seems to support the busy protocol, will adjust timeouts and set busy interval accordingly
[...]
Recv: ok P63 B63
Send: M113 S2
Recv: ok P63 B63
```

ThomasToka, 2023-03-29:
ok and when you now go back to the recent release..

does it still die on first homeing?

ThomasToka, 2023-03-29:
did you have a ubl firmware installed before you installed first of mines?

andre68723, 2023-03-29:
No, I directly went from latest stock `Ender-3 S1_Pro_HWv24S1_301_SWV2.0.8.26F4_FDM_LASER` to yours.
Before that, I had the stock it came with.

back to 
`firmware-20230326-220058-E3S1Pro-abl5x5-LA-F4.bin`
same problem
Recv: echo:Invalid mesh.

ThomasToka, 2023-03-29:
andre are you on discord or facebook? i have something to test for you.

ThomasToka, 2023-03-29:
i have found the initial problem of the g28 problem.

it was simple a wrong set brace in the last update.

i will push new bins in some minutes.

as this is a essential bugfix and i am ready with the folders and subfolders...

its folders and subfloders release then.

thanks for reporting.

ThomasToka, 2023-03-29:
i just pushed the update.

please redownload and check.

thank you for reporting and delivering data!

much appreciated!

ThomasToka, 2023-03-30:
i will close the issue. you can still feedback me here!

closed 2023-03-30 by ThomasToka

andre68723, 2023-03-31:
thanks for fixing. Verified it's working correct now.

1) Minor thought about ABL, the probe point (offset) is  45/45 from the left front edge. Any thoughts if that shold be rather outwards?
2) is M420 S1 required after G28 in the start code? I remember it's always enabled in the original fw -  #RESTORE_LEVELING_AFTER_G28

andre68723, 2023-03-31:
Addendum
3) maybe you should put the section from the commit msg (https://github.com/ThomasToka/MarlinFirmware/commit/e286acd969d370d77d959314ff7706328e9951b9) into the readme directly.  Q2 is mentioned there - "Restore Bedleveling after G28"

ThomasToka, 2023-03-31:
i just copy and paste what is wrote to 1) on facebook:

your printer can by design not reach more than the probe offset is on y min. there are two posibilieties to manage this:
1 not square, not centered mesh, meassuring 3 sides to the max reachable point, leaving one at about 42mm from the edge (y min).
2 square, centered mesh, with 45mm margin and extrapolation.
i have chosen 2) becaue the last 45 mm are extrapolated by the mesh routine.
extrapolation works great in this case, as all sides of the mesh are square.
makes more sense to have well measured mesh in the most printed areas.

for 2) no, not needed, as restore after g28 is compiled

3) will do with the next releases

thanks!