ThomasToka/MarlinFirmware #3: Leveling

beroliv, 2023-04-01:
Leveling stops at 98% with the 7x7 mesh on Ender 3 S1 pro

ThomasToka, 2023-04-01:
Try 5x5 please. Will verify ubl7x7 in some minutes.

ThomasToka, 2023-04-01:
An please show me your m503 output.

beroliv, 2023-04-01:
i level again and try to catch m503

ThomasToka, 2023-04-01:
yes please. mine is at 52% now ;)

ThomasToka, 2023-04-01:
Any upgrades like zero y offset mode or something like that?

beroliv, 2023-04-01:
no i dont know that something exist :-)

beroliv, 2023-04-01:
there is another problem but dont know to explain in english

beroliv, 2023-04-01:
there is no 503, the printer just stops and wait

ThomasToka, 2023-04-01:
i can confirm the 98%. I will try something.. give me 10 minutes.

m503 is the output of your setting sent through a terminal.

but as you dont have any modifictions and i can reproduce this i dont need the output then.

my tests started.

beroliv, 2023-04-01:
Try the ubl5x5

beroliv, 2023-04-01:
ocotoprint shows you the bed leveling even it is not finish

beroliv, 2023-04-01:
can we have a chat somewehere??

ThomasToka, 2023-04-01:
add me on facebook

beroliv, 2023-04-01:
i send you a msg, im not really a Fazebugger

ThomasToka, 2023-04-01:
5x5ubl works. i think it will have something to do with the final accounting of the percentage.

technically the ubl works.

ThomasToka, 2023-04-01:
The bug has been fixed and is a fixed version is available with the next bugfix release.

The bug is documented under the following shortcode in the changelog:

[MARLIN-E3S1FORK-BYTT-fix3] ubl7x7 finished percentage does not match and is stuck at 98%. With this release this visual bug is fixed.

I will let this open till the bugfix release has been pushed. Estimated releasetime 02.04.2023 12.00 CET.

ThomasToka, 2023-04-02:
[[MARLIN-E3S1FORK-BYTT-2]](https://github.com/ThomasToka/MarlinFirmware/commit/c15723371fc398cf65b7258edce4634ecf06ea32) 

has been released fixing this issue with [MARLIN-E3S1FORK-BYTT-fix3].

i will close this now.

thank you for finding and reporting this visual bug.

closed 2023-04-02 by ThomasToka