ThomasToka/MarlinFirmware #5: Firmware option with stock bed size

hugefishh, 2023-04-15:
Hi, @ThomasToka 
I'm really grateful someone trying to make that mess of s1 firmware into smth usable. As I can see you've did fantastic job!

I got it that you are not ready or don't want to release sources of your fixes right now.
But maybe its possible that you will release option with stock sized dimensions with print size "220 * 220 * 270" and X_MIN_POS -9, Y_MIN_POS -6 for s1pro?
The reason for that as I'm not really comfortable with wires stretch on the edges of extended print area for my S1 Pro
Also I'm using plastic part as small x-offset to prevent extruder wires from slightly crashing on left side during X homing

At the moment I'm using synman's firmware, but M600 issue is breaking my heart. I'd like to try yours but there is no way my wires will survive that kind of stretch for long.
Maybe you can build and post it temporarily, till you ready to release sources? 

Any way, thank you a lot for your time and effort you have put in here.

ThomasToka, 2023-04-15:
Good morning,
i will drop the source code 30 days after the release of [MARLIN-E3S1PROFORK-BYTT-4].
Expect it around the 9th of May.

As for your initial problem with the clearance: Use another holder for the cable. I have this one here:
https://www.thingiverse.com/thing:5537361
But turnerd around so the cable is on the right side of it.

With this your cable wont touch your frame on the left and on the right.

You are invited to join my discord for this project: https://discord.com/invite/Fh4jsUJWe6

ThomasToka, 2023-04-15:
And secondly: Even if i would change this as you wish your extruder will hit the frame with the cable as it has to reach the endstop position. So it wouldn´t change anything on the left side.

closed 2023-04-15 by ThomasToka

dl200010, 2023-06-09:
https://www.thingiverse.com/thing:5394194

I use this cable holder for my 3 S1 Pro. It moves the cable in front of the gantry. this keeps it from touching at all when homing the X axis. Just thought I would drop this here to help.