ThomasToka/MarlinFirmware #6: value of babystepping is probably not

0Locke0, 2023-04-15:
Hi,
after upgrading to your firmware "Marlin 2.1.x Ender3S1Pro-withLA-ubl5x5-F4"  i found a small issue by setting the value of z-Offset, whe you touch the up or donw button in the Leveling menue of the printer, it jumps very often higher than 0.01 for example from 3.01 to 3.11 after pushing again it goes from 3.11 to 3.04 and then it works for some pushes well than the same again.

Maybe it is dependent from the touchscreen?

ThomasToka, 2023-04-15:
This is the babystepping fix intrduced by Synman, which i slightly modified.

The firmware now corrects the value entered via touch screen to the "recorded right value" and adds then up.

Intended behavior and without this we would have skipped steps. And that is not cool.

Looks a bit strange. But yeah thats the fix ;)

ThomasToka, 2023-04-15:
You are invited to join my discord server: https://discord.com/invite/Fh4jsUJWe6

beroliv, 2023-04-15:
i would love it i can change the z offset also with a key pad. After flashing a new firmware i set via Terminal to my last known point. Tapping the Dsiplay for -3mm ist kind fustrating

ThomasToka, 2023-04-15:
as i dont have access to the sources of the screen there is not chance to have both active. someone will always complain about missing 0.01 or missing 0.05. 

so it is like it is. closing this. thanks for contributing to this fork!!!

closed 2023-04-15 by ThomasToka