ThomasToka/MarlinFirmware #7: Bed is not centered

3djc, 2023-05-02:
Printer: Creality Ender 3 S1 Pro

Slicer view:

[image]

Printed:
[image]

[image]

ThomasToka, 2023-05-02:
Please show me your M503, M115 and upload this gcode here.

Will have a look.

3djc, 2023-05-02:
I'm assuming you want the return of those two commands, but can I run them while printing ?

ThomasToka, 2023-05-02:
nope. connect with pronterface or any other terminal while not printing.

3djc, 2023-05-02:
ok, I'm using octoprint, so I have a terminal handy, but 4 hours 22 min printing left :)

ThomasToka, 2023-05-02:
[image]

thats what i see in cura when i load your gcode. so it simply is not centered in your creality slicer. 

however.. not a firmware problem.

3djc, 2023-05-02:
Understood and sorry for the bogus report.. Strangely, both Creality slicer and Cura give me the same error :( Any pointer on where that setting could be (both are default install)

ThomasToka, 2023-05-02:
probably you have your buildplate set to 220x220.

it is physical 235x235. so the firmware is set to physical dimensions.

change this and it should match.

3djc, 2023-05-02:
ok, will try that. Thanks a million (maybe worth adding as a side note to the installtion instructions ?)

ThomasToka, 2023-05-02:
no problem. all cool. enjoy the firmware. join my discord if you want..

https://discord.com/invite/Fh4jsUJWe6

closed 2023-05-02 by ThomasToka