ThomasToka/MarlinFirmware #9: Source code

jklap, 2023-05-16:
Hello @ThomasToka -- when do you plan to update your Marlin fork repo with the changes found in the firmware you have posted here?

ThomasToka, 2023-05-16:
When i am ready with the final release. Maybe 2 or 3 weeks.

ThomasToka, 2023-06-08:
The source code has been released today.

https://github.com/ThomasToka/Marlin/tree/MARLIN-E3S1PROFORK-BYTT

i have also made a pull request against marlin upstream.

https://github.com/MarlinFirmware/Marlin/pull/25948

closed 2023-06-08 by ThomasToka

mikep1998, 2023-06-08:
Hello, Is there a release of the final version or just RC4?

Thanks
Mike

https://github.com/ThomasToka/Marlin/tree/MARLIN-E3S1PROFORK-BYTT

MarlinFirmware/Marlin#25948<https://github.com/MarlinFirmware/Marlin/pull/25948>

ThomasToka, 2023-06-08:
Binary release is tomorrow morning.