ThomasToka/MarlinFirmware #11: Hotend offset

LVianaS2, 2023-05-25:
Hello friend! I changed the hotend offset (X-55, Y0), and now I can't access the points on the right side of the bed for manual leveling. 
Can you tell me if it's a firmware issue?

I would like to know if it is possible to adjust the mesh of the table in your firmware. Even with the automatic leveling, some points on the table are still uneven.

Tanks!

ThomasToka, 2023-05-25:
0 Y offset is not supported. Sorry. It has todo with other variables like probe margin and home offsets.

If you bed is that bowed i cant help you. Change it. Sorry.

LVianaS2, 2023-05-25:
I'm using this modification (https://www.printables.com/model/259877-ender-s1-sprite-extruder-cr-touch-mount) to align the CR-touch to the hotend nozzle in an attempt to improve mesh leveling.

ThomasToka, 2023-05-25:
May be. Sorry for you. Not supported. And yes in my opinion not needed so i wont support this for now.

closed 2023-05-25 by ThomasToka