ThomasToka/MarlinFirmware #12: Touchscreen is freezing

LVianaS2, 2023-05-28:
Hello, the touchscreen is freezing on the leveling page. When I touch the buttons, it makes a beeping sound, but it doesn't respond to commands.

ThomasToka, 2023-05-28:
Cant reproduce. Works as intended.

Please post on my discord in the #bugreport channel with a short video where it exactly happens.

I have only seen something with cable bricks that accassionally accure and lead to inconsitent results.

ThomasToka, 2023-05-28:
And: Are you sure both screen and mainboard updated to rc3 ?

LVianaS2, 2023-05-28:
> And: Are you sure both screen and mainboard updated to rc3 ?

Yes, I updated both firmware, both on the screen and on the motherboard.

LVianaS2, 2023-05-28:
I added the video to your discord

ThomasToka, 2023-05-28:
After intersecting you complain:

- there is no freeze
- the button only reacts if the steppers were deactivated
- as entering leveling page automatically does a G28 the steppers are in a "valid known position".

so everything works as intended.

to testwise activate the button:

- send M84 over a terminal or sender
- push the home button
- axes will be homed once and are back in a "valid and known position"
- after this the button will only beep again till the stepper position is again unvalid 
- the beeping can be supressed.

so: nice try with a bug. but it all works like intended. a bit unlucky that i cant suppress the beep, but thats how it is.

closed 2023-05-28 by ThomasToka