ThomasToka/MarlinFirmware #13: No filament extrusion after restarting paused print

maeldupontMX10, 2023-06-06:
Hello,

I'am on Ender 3 S1 Plus. I installed your firmware on the 4th of June. It tried pausing a print the clean it a little bit but when I restarted the print, the head was moving following the Gcode but there was no more filament extrusion. The visible gear of the sprite extruder was not spining anymore. It did paused prints in the past with the stock firmware without issues.

Firmware S1 PLUS F4 ABL 5x5

Thank you

ThomasToka, 2023-06-06:
Hello,

how you started the print? SD print or Host print from for example Octorpint?
How you paused the print? Over touch or over host?
How you resumed print? Ober touch or over host?

ThomasToka, 2023-06-06:
Sorry to say but i cant reproduce this. Exactly this works here.

If it is reproducable please make a short video of it.

ThomasToka, 2023-06-06:
and what i have seen: as the cable goes to park position, users having a brick already but not noticed yet.. notice this on the park position cause the ribbon is bowed there most and so the brick comes into play..

you can also come to my discord. there are many users sharing thereir stuff..

closed 2023-06-06 by maeldupontMX10

maeldupontMX10, 2023-06-06:
Yes this explain why it did not extrude, than you ;)