ThomasToka/MarlinFirmware #16: autorun crtouch

fabiociconello, 2023-06-29:
Olá Tomas.

Primeiramente obrigado pelo firmware, isso ajudou muito a comunidade, continue esse bom trabalho.

Minha duvida é qual o objetivo do "Autorun do CRTouch", para que serve?

Obrigado

ThomasToka, 2023-06-29:
Sorry i only can support in english.

It seams you are asking about the autorun on the cr touch site. it measures your tramming points automatically. they should all be nearly the same to have a flat surface. 

please don’t use github issues for simple questions. you can join my discord for simple questions.

closed 2023-06-29 by ThomasToka

fabiociconello, 2023-06-30:
Hi Tomas

How can I join with you on discord?

Thanks and I'm sorry to use the git hub issues

ThomasToka, 2023-06-30:
https://discord.com/invite/Fh4jsUJWe6