ThomasToka/MarlinFirmware #18: UBL100 fails after point 90

lalalamoneyright, 2023-08-14:
UBL100 fails after point 90 every time, tried M501 M502, power  cycling and repeating, checked the manual leveling. 
Pronterface gives nothing but Mesh calibration failed. 
It worked first time with no issue, no other SW changes done since then. 
[image]

ThomasToka, 2023-08-14:
second run also was successfull. so most probably not a firmware problem.

lalalamoneyright, 2023-08-14:
[image]

Just a no Y offset probe mount, maybe this could influence?

ThomasToka, 2023-08-14:
yes. thats it. it wont work reliable with a 0 y offset mod.

ThomasToka, 2023-08-14:
Show me your M503 please.

lalalamoneyright, 2023-08-14:
I'm not near my printer rn but the offset is X46.5 Y0 that's what I changed in the config. I'll install the stock CR mount later and redo the meshing.

closed 2023-08-15 by ThomasToka

lin-ycv, 2023-09-11:
FYI 7x7 ubl also fails with 0 offset mod

ThomasToka, 2023-09-11:
> FYI 7x7 ubl also fails with 0 offset mod

maybe. but its not supported. so i wont invest any time in it. sorry.

ThomasToka, 2023-11-03:
please try the v016 builds.

i have introduced dynamic margins and mesh switch to ubl also now.

https://github.com/ThomasToka/MarlinFirmware/releases/tag/MARLIN-E3S1PROFORK-BYTT-v016