ThomasToka/MarlinFirmware #20: Update for both Display and Mainboard does not work

KrOnAsK, 2023-08-17:
Hi, I tried to update my Ender 3 S1Plus today and followed your steps exactly but when I try to update the display firmware the screen doesnt show anything for 10min exept for the backlight that is on (and the fan so I know the printer is actually on) and then after that I tried updating the Mainboard and the printer just booted like normal. I put the bin file in the folder which is correctly named and formated to fat32 with a size of 4096 and the size of the Volume is 2048 just like you described. 

This is the back of my Display
[image]
and this is the  Mainboard
[image]

ThomasToka, 2023-08-17:
ok the content of the cards looks correct.

please try with another card the same way you did with this card.

i have noticed that some cards simply dont work. one cant say if cheap or expensive.. for me the cheapest work best.

KrOnAsK, 2023-08-17:
Okay with a different SD card it worked like a charm, thanks. Now only one more question since the wiki isnt really populated yet, where can I find more information about what I am able to do with this firmware and what all the functions do. Also is Host Action Commands enabled?