ThomasToka/MarlinFirmware #22: is M420 S1 required for UBL?

lin-ycv, 2023-09-06:
as the title, does the firmware automatically apply the mesh? or do I have to set the command M420 S1?
I don't really see a difference between setting and not setting M420, but then again my bed is trammed near perfectly.

ThomasToka, 2023-09-06:
The mesh is save automatically.

M420 S1 restores the state of leveling before G28. So i would recommend it after G28 in Start gcode.

closed 2023-09-06 by ThomasToka

lin-ycv, 2023-09-08:
@ThomasToka just wondering if you could provide a sample start gcode that you use for ubl, when I print I see high and low spots that correspond to the bed mesh values (Where there's high points on the bed mesh, the print becomes squashed), which seems to indicate that the machine isn't using the bed mesh

This is what I currently have:
```gcode
M140 S60
M105
M190 S60
M104 S200
M105
M109 S200
M82 ;absolute extrusion mode
; Ender 3 S1 Pro Start G-code
G28 ;Home
M420 S1
G92 E0 ;Reset Extruder
G1 Z2.0 F3000 ;Move Z Axis up
G1 X10.4 Y200.0 Z0.28 F5000.0 ;Move to side a little
G1 X10.4 Y20 Z0.28 F1500.0 E30 ;Draw the second line
G92 E0 ;Reset Extruder
G1 Z2.0 F3000 ;Move Z Axis up
G92 E0
G92 E0
G1 F2400 E-0.8
```

I've also tested some G29 commands, but haven't seen any real difference, thanks.

ThomasToka, 2023-09-08:
You should check if you mesh is enabled and visible.

Send for example "M420 V" after bedleveling.

lin-ycv, 2023-09-08:
this is what I get with `M420 V`, When I print i'm getting squish on the back and right.
```
Connecting...
Printer is now online.
>>> M420 V
SENDING:M420 V
Bed Topography Report:
    ( 27,190)                      (208,190)
        0       1       2       3       4
 4 | -0.390  -0.448  -0.083  -0.359  -0.342
   |
 3 | -0.119  -0.229  -0.272  -0.220  -0.165
   |
 2 | +0.046  -0.150  -0.183  -0.120  -0.046
   |
 1 | -0.025  -0.128  -0.163  -0.089  -0.011
   |
 0 | +0.041  -0.079  -0.107  -0.017  +0.103
        0       1       2       3       4
    ( 27, 27)                      (208, 27)
Mesh is valid
Storage slot: 0
echo:Bed Leveling ON
echo:Fade Height 5.00
ok P15 B7
```
When printing, it says it's using the bed mesh
```
echo:SD card ok
echo:Now fresh file: CE3S1P~1.GCO
File opened: CE3S1P~1.GCO Size: 92479
File selected
echo:enqueueing "M23 ce3s1p~1.gco"
echo:enqueueing "M24"
echo:Now fresh file: ce3s1p~1.gco
File opened: ce3s1p~1.gco Size: 92479
File selected
//action:resume
echo:Bed Leveling ON
echo:Fade Height 5.00
```

ThomasToka, 2023-09-08:
your bed is simply not good enough leveled.

do cr-touch meassuring under aux.lvl -> cr-touch -> autorun then adjust your points and run it again till the four outer points are nearly the same.

always hot, and after waiting 10 minutes till you bed expanded fully.

lin-ycv, 2023-09-08:
> your bed is simply not good enough leveled.

How close does the 4 points need to be? I have multiple S1, some of them have super flat beds, but some are warpped (0.2 delta between highest and lowest point) and basically impossible to get them nearly the same.

ThomasToka, 2023-09-08:
0.2 is a whole layer and too much. i get my within 0.05.

lin-ycv, 2023-09-08:
what would you recommend for warpped beds? this one for example, have all the for corners near perfect when using the aux level, but the bed mesh has a delta of 0.182 (waited 20 min after heating stabilized, made sure there's no filaments stuck under the bed)

Aux (Delta 0.01):
```
Bed X:45.0 Y:45.0 Z:-0.018
Bed X:190.0 Y:45.0 Z:-0.013
Bed X:45.0 Y:190.0 Z:-0.013
Bed X:190.0 Y:190.0 Z:-0.023
Bed X:117.5 Y:117.5 Z:-0.018
```

Bed Mesh (delta 0.182)
```
        0       1       2       3       4
 4 | -0.013  +0.024  -0.002  +0.000  +0.084
   |
 3 | +0.052  +0.075  -0.003  -0.011  +0.062
   |
 2 | +0.019  +0.050 [-0.013] -0.008  +0.082
   |
 1 | +0.002  +0.048  -0.013  -0.025  +0.089
   |
 0 | +0.012  +0.072  +0.036  +0.043  +0.157
 ```