ThomasToka/MarlinFirmware #23: [bug] autorun and set offset

lin-ycv, 2023-09-11:
i've noticed 2 bugs in the latest version
1. during cr-touch aux level, if you accidentally tap autorun more than one, it won't be able to do autorun until a reboot
2. hotend offset can only be set when booted, once you home the printer, the offset adjustment moves the hotend instead of changing the offset.

ThomasToka, 2023-09-11:
1) will be fixed in v009
2) dont understand explain better.

lin-ycv, 2023-09-11:
for setting the hotend offset
Scenario 1 (working as expected):
1. boot up the machine
2. Settings > Adv Set >offsets > hotend offsets
3. adjust values (hot end doesn't move)
4. exit the screen and offset values are set

Scenario 2 (not working):
1. boot up the machine
2. home the machine
3. Settings > Adv Set >offsets > hotend offsets
4. adjust values (hot end moves)
5. exit the screen and offset values aren't set

ThomasToka, 2023-09-11:
> for setting the hotend offset Scenario 1 (working as expected):
>
>     1. boot up the machine
>
>     2. Settings > Adv Set >offsets > hotend offsets
>
>     3. adjust values (hot end doesn't move)
>
>     4. [...]

i just tried to reproduce this.

this function uses babysteps.

babysteps are only added to the movements if the head is in safe position after G28.

before G28 the value is added and on next g28 it gets active.

- booted printer
- went directly to hotend offsets
- set hotend offset for x from -31.5 to -51.50
- left the site
- sent m503 with pronterface.
result:

```
echo:  M851 Probe Offset X-51.50 Y-41.80 Z-1.90 ; (mm)
```

if i now home my printer with G28 the change is reflected.

did i miss something?

closed 2023-09-13 by ThomasToka