ThomasToka/MarlinFirmware #26: Preview does not work in OrcaSlicer

cybertino, 2023-09-20:
Looks like Orca generates thumbnail in PNG format which does not seem to be supported in Creality Ender 3 S1 Pro.

cybertino, 2023-09-20:
A workaround for this could be:

- decode base64 of the thumbnail into bytes
- load bytes into Pillow instance and convert it into JPEG
- encode JPEG into base64

The main inconvenience (apart from being slower) is that you would need to install the [Pillow package](https://pillow.readthedocs.io/):

```
pip install Pillow
```

ThomasToka, 2023-09-20:
[image]

the problem is more general i think. can you generate at all with it?

ThomasToka, 2023-09-20:
ah me stupid. i had a typo in my file.

I will now have a look regarding png jpg change..

ThomasToka, 2023-09-21:
I have developed a matching script, updated the script in the repo and updated the instructions in the wiki.

thanks for reporting.

closed 2023-09-21 by ThomasToka

cybertino, 2023-09-22:
Awesome!

Thanks for fixing @ThomasToka!

lin-ycv, 2024-01-08:
Question, Orca (1.9) can output JPG format, but it doesn't seems to work with the firmware
what's the difference between the jpg of Orca and jpg of the script?