ThomasToka/MarlinFirmware #30: Gibt es eine Installationsanleitung

Duffy987, 2023-10-06:
Entschuldigung das ich es über Issue mache aber ich komme nicht anders weiter.
1. Ist mein Englisch nicht gut
2. Finde ich mich nach Monaten in dem Github Verzeichnissen nicht zurecht.

Deshalb meine Frage, gibt es irgendwo eine Erklärung (Anleitung) wo man die Software herunterladen kann, ja ich weis bei Github, nur wo und wie geht das Installationsprozedere von statten geht.

Ich habe auch in 3D Foren keine Hilfe gefunden.

Mir ist schon einmal ein 3D Drucker gestorben das möchte ich eigentlich vermeiden.

Über eine Hilfe würde ich mich freuen, und nochmals Entschuldigung das ich diesen Weg gewählt habe vielleicht gibt es ja andere die auch dieses Problem haben.

Google:

Sorry for doing it via Issue but I can't get any further.
1. My English is not good
2. After months I can't find my way around the Github directories.

That's why my question is, is there an explanation (instructions) somewhere where you can download the software, yes I know from Github, just where and how the installation procedure works.

I didn't find any help in 3D forums either.

I've had a 3D printer die before and I actually want to avoid that.

I would be happy about some help, and I apologize again for choosing this route, maybe there are others who also have this problem.

ThomasToka, 2023-10-06:
hello,

https://www.youtube.com/watch?v=bfDjoWGbl44

https://github.com/ThomasToka/MarlinFirmware/wiki/Installation-%E2%80%90-mainboard-firmware

https://github.com/ThomasToka/MarlinFirmware/wiki/Installation-%E2%80%90-screen-firmware

https://github.com/ThomasToka/MarlinFirmware/releases/download/MARLIN-E3S1PROFORK-BYTT-v012/installation.txt

english only. sorry.

closed 2023-10-06 by ThomasToka

Duffy987, 2023-10-06:
Thank youShould i delete the Issue again?Regards duffy. Gesendet mit der mobilen Mail AppAm 06.10.23 um 11:01 schrieb ThomasToka
                
            
            
                
                    Von: "ThomasToka" ***@***.***>Datum: 6. Oktober 2023An: "ThomasToka/MarlinFirmware" ***@***.***>Cc: "Author" ***@***.***>,"Duffy987" ***@***.***>Betreff: Re: [ThomasToka/MarlinFirmware] Gibt es eine Installationsanleitung (Issue #30)
                
                
hello,
https://www.youtube.com/watch?v=bfDjoWGbl44
https://github.com/ThomasToka/MarlinFirmware/wiki/Installation-%E2%80%90-mainboard-firmware
https://github.com/ThomasToka/MarlinFirmware/wiki/Installation-%E2%80%90-screen-firmware
https://github.com/ThomasToka/MarlinFirmware/releases/download/MARLIN-E3S1PROFORK-BYTT-v012/installation.txt
english only. sorry.

Duffy987, 2023-10-06:
Ok i have Seen its closed. 👍Gesendet mit der mobilen Mail AppAm 06.10.23 um 11:01 schrieb ThomasToka
                
            
            
                
                    Von: "ThomasToka" ***@***.***>Datum: 6. Oktober 2023An: "ThomasToka/MarlinFirmware" ***@***.***>Cc: "Duffy987" ***@***.***>,"Author" ***@***.***>Betreff: Re: [ThomasToka/MarlinFirmware] Gibt es eine Installationsanleitung (Issue #30)
                
                
hello,
https://www.youtube.com/watch?v=bfDjoWGbl44
https://github.com/ThomasToka/MarlinFirmware/wiki/Installation-%E2%80%90-mainboard-firmware
https://github.com/ThomasToka/MarlinFirmware/wiki/Installation-%E2%80%90-screen-firmware
https://github.com/ThomasToka/MarlinFirmware/releases/download/MARLIN-E3S1PROFORK-BYTT-v012/installation.txt
english only. sorry.

Duffy987, 2023-10-12:
Hello Thomas, 

 

Thanks for your link: 

https://github.com/ThomasToka/MarlinFirmware/blob/Firmware-Binaries/Ender-3S1_Pro_HWv24S1_301_SWV2.0.8.28F4_F401_FDM_LASER.zip

I think that's the original link from Creality. The firmware update didn't work once. 

 

So a question, first the monitor and then the motherboard? In Creality's YouTube video it's the other way around. 

 

In your Installation.txt you also describe first monitor then motherboard. 

 

I'm too worried about breaking it so I probably won't do it again today.

 

Many greetings Duffy

 

 

 

 

 

 

https://github.com/ThomasToka/MarlinFirmware/blob/Firmware-Binaries/Ender-3S1_Pro_HWv24S1_301_SWV2.0.8.28F4_F401_FDM_LASER.zip

hello,

https://www.youtube.com/watch?v=bfDjoWGbl44

https://github.com/ThomasToka/MarlinFirmware/wiki/Installation-%E2%80%90-mainboard-firmware

https://github.com/ThomasToka/MarlinFirmware/wiki/Installation-%E2%80%90-screen-firmware

https://github.com/ThomasToka/MarlinFirmware/releases/download/MARLIN-E3S1PROFORK-BYTT-v012/installation.txt

english only. sorry.

Duffy987, 2023-10-14:
Is it not possible to send a donation via PayPal?

Regards Duffy

 

https://www.youtube.com/watch?v=bfDjoWGbl44

https://github.com/ThomasToka/MarlinFirmware/wiki/Installation-%E2%80%90-mainboard-firmware

https://github.com/ThomasToka/MarlinFirmware/wiki/Installation-%E2%80%90-screen-firmware

https://github.com/ThomasToka/MarlinFirmware/releases/download/MARLIN-E3S1PROFORK-BYTT-v012/installation.txt

ThomasToka, 2023-10-14:
hello,
sorry. but you can use github sponsors or patreon:

https://github.com/sponsors/thomastoka
https://www.patreon.com/ThomasToka

thank you in advance!

ThomasToka, 2023-10-14:
I have added paypal in the menu on the right side with my paypal.me link.

https://github.com/ThomasToka/MarlinFirmware

on the right side under "Sponsor this project"

Duffy987, 2023-10-14:
That's a good idea. Thank you for your effort.

 

Regards Udo aka Duffy

 

https://github.com/ThomasToka/MarlinFirmware

Duffy987, 2023-10-30:
Hello Thomas, I need your help outside of 3D printing please. You probably know your way around Github. Can you please tell me where the download link for: 

 

https://github.com/search?q=ZigStarGW-MT <https://github.com/search?q=ZigStarGW-MT&type=repositories> &type=repositories

 

Can be found? 

 

Best regards, Duffy