ThomasToka/MarlinFirmware #31: UBL probe not probing whole bed (right side) X max 210

DjordjeMandic, 2023-10-11:
Ender 3 S1 Pro

X-45 Y0 CRTouch mount

X can actually go up to 256

So for X210 probe actual nozzle position has to be at X255 which is possible 

Firmware [MARLIN-E3S1PROFORK-BYTT-v013]

```
Probing mesh point 91/100.
//action:notification Probing Point 91/100
Bed Topography Report:
    ( 25,190)                                                              (210,190)
        0       1       2       3       4       5       6       7       8       9
 9 | -0.007  -0.020  -0.025  -0.002  +0.002  -0.018  +0.045  +0.052  +0.060    .
   |
 8 | -0.002  -0.022  -0.020  -0.018  +0.015  -0.027  +0.045  +0.045  +0.047    .
   |
 7 | -0.005  -0.010  -0.010  +0.007  +0.029  +0.020  +0.062  +0.050  +0.055    .
   |
 6 | +0.025   0.000  -0.007  +0.016  +0.032  +0.027  +0.060  +0.067  +0.075    .
   |
 5 | +0.015  -0.012  -0.007  +0.020  +0.015  +0.018  +0.040 [+0.062] +0.080    .
   |
 4 | +0.025   0.000  +0.022  +0.027  +0.025  +0.007  +0.050  +0.075  +0.100    .
   |
 3 | +0.024  -0.002  +0.007  +0.015  +0.005  -0.022  +0.032  +0.052  +0.085    .
   |
 2 | +0.025  -0.007  -0.005  +0.010  -0.005  -0.012  +0.040  +0.065  +0.076    .
   |
 1 | +0.057  +0.045  +0.022  +0.035  +0.045  +0.050  +0.095  +0.099  +0.097    .
   |
 0 | +0.105  +0.072  +0.047  +0.070  +0.065  +0.062  +0.112  +0.112  +0.121    .
        0       1       2       3       4       5       6       7       8       9
    ( 25, 25)                                                              (210, 25)
Mesh saved in slot 0
Done.
echo:Invalid mesh.
Error:Failed to enable Bed Leveling
Error:Failed to enable Bed Leveling

echo:Bed Leveling OFF
echo:Fade Height 5.00
echo:Settings Stored (383 bytes; crc 25756)
Mesh saved in slot 0
//action:notification Settings Stored
//action:notification Settings Stored
//action:prompt_end
//action:prompt_begin M1 Stop
//action:prompt_button Continue
//action:prompt_show
echo:; Linear Units:
echo:  G21 ; (mm)
echo:; Temperature Units:
echo:  M149 C ; Units in Celsius
echo:; Filament settings (Disabled):
echo:  M200 S0 D1.75
echo:; Steps per unit:
echo:  M92 X79.50 Y79.50 Z400.00 E424.90
echo:; Max feedrates (units/s):
echo:  M203 X1000.00 Y1000.00 Z10.00 E50.00
echo:; Max Acceleration (units/s2):
echo:  M201 X1000.00 Y1000.00 Z100.00 E1000.00
echo:; Acceleration (units/s2) (P<print-accel> R<retract-accel> T<travel-accel>):
echo:  M204 P1000.00 R1000.00 T1500.00
echo:; Advanced (B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_jerk> Y<max_jerk> Z<max_jerk> E<max_jerk>):
echo:  M205 B50000.00 S0.00 T0.00 X12.00 Y12.00 Z0.40 E5.00
echo:; Home offset:
echo:  M206 X-3.50 Y-2.00 Z0.00
echo:; Flow Percentage:
echo:  M221 S100
echo:; Unified Bed Leveling:
echo:  M420 S0 Z5.00 ; Leveling OFF
Unified Bed Leveling System v1.01 inactive
echo:Active Mesh Slot 0
echo:EEPROM can hold 3 meshes.
echo:; Material heatup parameters:
echo:  M145 S0 H200.00 B60.00 F255
echo:  M145 S1 H240.00 B100.00 F255
echo:  M145 S2 H230.00 B80.00 F128
echo:  M145 S3 H190.00 B50.00 F128
echo:; Hotend PID:
echo:  M301 P20.13 I1.86 D54.35
echo:; Bed PID:
echo:  M304 P127.12 I23.03 D476.81
echo:; Power-loss recovery:
echo:  M413 S0 ; OFF
echo:; Z-Probe Offset:
echo:  M851 Probe Offset X-45.00 Y0.00 Z0.00 ; (mm)
echo:; Input Shaping:
echo:  M593 X F40.00 D0.15
echo:  M593 Y F40.00 D0.15
echo:; Linear Advance:
echo:  M900 K0.05
echo:; Filament load/unload:
echo:  M603 L0.00 U60.00 ; (mm)
echo:; Filament runout sensor:
echo:  M412 S1 ; Sensor ON
```

ThomasToka, 2023-10-11:
everything like planned. 

if you want to change this you could download the latest sources and compile yourself for your needs and change naturally all other dependant things like the screen wie dgus software.

and no a stock printer can not go on x to 256 but to 242. Only when you modify the cable to the front this is possible.

officially 0 y offset mod is not supprted cause not needed in my opinion.

sorry. there wont be anthing for you released by me in this case.

ThomasToka, 2023-10-11:
and read this also: https://github.com/ThomasToka/MarlinFirmware/wiki/The-x-and-y-margins

closed 2023-10-11 by ThomasToka

DjordjeMandic, 2023-10-11:
Whats discord link because https://discord.com/invite/Fh4jsUJWe6 is not working

ThomasToka, 2023-10-11:
the link still works. we have daily new users.

DjordjeMandic, 2023-10-11:
[image]

DjordjeMandic, 2023-10-11:
Fixed, browser problem

DjordjeMandic, 2023-10-11:
duplicate of #18

ThomasToka, 2023-10-11:
> duplicate of #18

no this is simply wrong. this is not a duplicate of #18 

this was about non working probing what is not more the case as the offsets are accounted while changing margins.

ThomasToka, 2023-11-03:
please try the v016 builds.

i have introduced dynamic margins and mesh switch to ubl also now.

https://github.com/ThomasToka/MarlinFirmware/releases/tag/MARLIN-E3S1PROFORK-BYTT-v016