ThomasToka/MarlinFirmware #35: Some buttons on the touch screen (such as Leveling) don't do anything

Berzeger, 2023-11-10:
Hi, in the latest firmware (v016) on my S1 Plus I cannot access the leveling page through the touchscreen. The display beeps when I touch it, but nothing else happens. I also can't access the settings where you should be able to turn the sounds on and off (the same problem as with the leveling).
I was previously using synman's firmware.
This was my process:

1. While still having synman's firmware installed, I upgraded my screen firmware to the official 1.0.3.
2. I then upgraded the screen firmware using Ender-3S1_Pro_HighSpeed_E3S1PROFORKBYTT_Pro_Plus_v016.
3. After that, I installed firmware-20231104-175341-E3S1Plus-abl-v016-F4.bin.
4. I verified everything in the About screen on my printer.

If I manually run G28 and then G29, the display successfully enters the leveling menu and then immediately begins the autoleveling process, but it's still annoying. Note that while in this menu, the CR-touch icon on the display suffers from the same issue - it beeps, but does nothing else.

Is this just my issue, or is something wrong with the firmware? Note that I haven't used any previous versions (I've just discovered Thomas' firmware and decided to give it a go since it is being actively maintained).

Thanks!

ThomasToka, 2023-11-10:
Hello, look at the about page. You should see in the Screen FW field also my firmware.
Please show a pic of your about page.

ThomasToka, 2023-11-10:
You need to install my mainboard firmware also.

Berzeger, 2023-11-10:
That's what I thought I did in step 3, but I guess something went wrong. Sorry about that. I'll put the SD card in the printer once more and will get back to you with results.

ThomasToka, 2023-11-10:
No Problem. I am here to Support you. We have also a grat discord Server with a bunch of users. Link is in the installation.txt.

Berzeger, 2023-11-10:
I did the process again and it works and looks awesome. I don't know why the printer didn't install it the first time, but I should have checked more thoroughly instead of creating a GitHub ticket. Thank you very much and I apologize again... I guess having a 2 year old crying through the night doesn't help my cognitive processes. :)
I'll check out the Discord server!

closed 2023-11-10 by Berzeger