ThomasToka/MarlinFirmware #36: changing filament

Romarioo7, 2023-11-13:
Hello my friend! thanks for the firmware. I can't seem to print with changing colors in the slicer. Tell me how to properly configure the slicer to pause and replace the filament. Thank you in advance!

ThomasToka, 2023-11-13:
Hello,
this is a "issue" reporting system for reporting problems with the firmware.

This a not a support channel for basic printing/feature questions.

Please use the channels on the discord for such questions.

To you question:

a) you can use M600 filament change:

[image]

b) you can use M0 "pause at height"

https://github.com/ThomasToka/MarlinFirmware/wiki/Pause-at-height-(M0)

closed 2023-11-13 by ThomasToka

ar360yandexru, 2023-11-27:
Good afternoon, can you  add the option to change the filament next to the pause button.
For example, I paused -> pressed the "change filament" button-> drove off to the XY parking place -> replaced the filament -> squeezed out 100mm of filament so that the replacement took place -> removed it carefully with my hand -> pressed the button -> continued printing.

Also, this algorithm can be applied as soon as the filament motion sensor is triggered

Thank you very much from Russia, Samara.