ThomasToka/MarlinFirmware #37: Trouble with screenfirmware

J-Tevop, 2023-11-17:
Hi,

No idea if it should post it here, sorry in advance. I am a little lost on the screen firmware.

Mainframe is working perfectly fine. The screen does not seem to update.

[image]

[image]

After following your instructions the firmware version doesn't update. Any ideas?

ThomasToka, 2023-11-17:
Hello,
Your card seems not to be formatted correctly.
Please have a look im the wiki.
https://github.com/ThomasToka/MarlinFirmware/wiki/Installation-%E2%80%90-screen-firmware

ThomasToka, 2023-11-17:
You can also join the discord for Support.

J-Tevop, 2023-11-17:
Hi,

Thanks for the quick response.

I just did that. No difference, unfortunately...

ThomasToka, 2023-11-17:
Did you Partition your sd card to 2gb like describe in the wiki?

J-Tevop, 2023-11-17:
Yep!

Where is the discord. Willing to join :)

ThomasToka, 2023-11-17:
Look in the Installation.txt

closed 2023-11-17 by ThomasToka