ThomasToka/MarlinFirmware #39: Continue printing after power outage.

KosolapbIu, 2023-11-22:
Hello. After updating the firmware, printing does not continue when the power is turned off... Is there something wrong with me or does the firmware not support continued printing? Or maybe this needs to be enabled somehow? Thank you. Have a good day...

ThomasToka, 2023-11-22:
Hello,

did you enable it either with 

```
M413 S1
M500
```

or on the main site the upper right toggle icon?

[image]

Default it is not enabled.

KosolapbIu, 2023-11-22:
Oh... it was "Off". Now I turn this to "On"... Thanks. I'm just a beginner in 3d printing. Good day!!!

closed 2023-11-22 by KosolapbIu

ThomasToka, 2023-11-22:
No problem.

You can also join out discord https://discord.com/invite/Fh4jsUJWe6 .

Have a sunny day.