ThomasToka/MarlinFirmware #40: Size Bed

isaiasemanuel22, 2023-12-08:
https://github.com/ThomasToka/MarlinFirmware/blob/733d72c93c891f306565abcee0f572ea56868123/LinearAdvance_Ender3S1Pro_E3S1PROFORKBYTT_kfactor.gcode#L22C19-L22C20

Buenas , primero felicitarte y agradecerte por los hermosos firware que estas entregando. 
He notado que cuando quiero hacer impresiones a los extremos , las impresiones se salen de la cama de impresion , y es porque la Ender 3 s1 pro , es de 220mm x 220mm no supe como corrergirlo hasta que vi tu codigo. Y ahora que lo  veo nose como solucionarlo exactamente, pero logre encontrar el error.

Muchas gracias!

ThomasToka, 2023-12-08:
Sorry. English please.

The bed size is 235x235 for the Ender 3 S1 Pro.

isaiasemanuel22, 2023-12-08:
Okay, my Ender 3 s1 pro has a 220 x 220 bed, how can I edit this in the firmware?

isaiasemanuel22, 2023-12-08:
because you have many things that I love, functionalities and screens, but I can't fit the prints on the bed well due to size differences

ThomasToka, 2023-12-08:
you are wrong. your ender 3 s1 pro has 235x235. measure it.

this here is not a support forum for simple basics like this.

first of all: read the installation.txt completely! this is covered inside of it.

- adjust your home offsets accordingly
- adjust your slicer bed size to the correct 235x235
- adjust your purge lines accordingly

the solution is not to set the firmware to a wrong value. you need to calibrate your printer correctly. simple as that.

again: read the installation.txt. all is covered in there.

if you need furter support come to the discord. the link is also in the installation.txt.

i will close this here as this is invalid and basically wrong on your site.

closed 2023-12-08 by ThomasToka