ThomasToka/MarlinFirmware #41: S1 Pro Bed Leveling UI Incorrect

DubStepMad, 2023-12-10:
Hi, just used this firmware and noticed on the touch screen when leveing that position 5 and 4 are switched around.

Looking straight ahead from the printer.
- Point 5 should be the back left but its the back right.
- Point 4 should be the back right but its the back left.

ThomasToka, 2023-12-10:
Hello,
please show me a picture of the about site of your printer.

closed 2023-12-11 by ThomasToka

DubStepMad, 2024-01-09:
@ThomasToka sorry for the delay, heres a video instead as it shows the bug more: https://drive.google.com/file/d/1aoNzQetP2o7qnLgPx80r9z0I9QXNyxKZ/view?usp=drive_link

ThomasToka, 2024-01-09:
[image]

this is the same site in my firmware.

DubStepMad, 2024-01-13:
> Again: Please send me a picture of the about site of your printer.
> 
> It is clearly visible that you dont have my screen installed.
> 
> Naturally this does not work.
> 
> Please join my discord and seek for help in #installation-help .

Can you link your discord so I can join as its not listed on your github profile or the repo.