ThomasToka/MarlinFirmware #42: Extruder fan not doing anything

legros04, 2023-12-11:
Hey Thomas having an issue where I get nothing out of the extruder cooler fan. the part cooler fan works fine but not the extruder cooler.

ThomasToka, 2023-12-11:
Noone has thi. So Probably Hardware.

Please show a picture of the about site of your printer.

legros04, 2023-12-11:
this is a brand new printer arrived yesterday, i also swapped out the sprite extruder for a spare i have and the same issue happend.

ThomasToka, 2023-12-11:
Please show the picture.

legros04, 2023-12-11:
[image]
[image]
[image]

legros04, 2023-12-11:
both extruders are new and both work fan 2 only after firmware upgrade  (i swapped the fan 1 and 2 plugs and the extruder fan works when plugged into fan 2 but then the part cooler fan does not work)

ThomasToka, 2023-12-11:
Of the about site of your printer Please.

legros04, 2023-12-11:
site? sorry i do not follow

legros04, 2023-12-11:
[image]

legros04, 2023-12-11:
sorry thought you said side

legros04, 2023-12-11:
ill go back to stock and see what happens should i go back to stock screen fw as well?

ThomasToka, 2023-12-11:
Yes you need to.

legros04, 2023-12-11:
ended up doing a bunch of digging and found [this](https://www.youtube.com/watch?v=Ijxh1nnUGlA&t=82s) apperently you can fry the ppm output by plugging in the laser module with the machine on. (of course i did that) and now i have no ppm to control the fan hence the fan not working.) at least i think a new board is on its way right now and i hope that is the fix. 

ps. thanks for the quick response really appreciate it

ThomasToka, 2023-12-11:
autch. no problem. good luck and a speedy recovery.

closed 2023-12-11 by ThomasToka

legros04, 2023-12-11:
thanks, also im not as well versed with these boards as you are would ppm issues stop the fan like that?

ThomasToka, 2023-12-11:
please come over to the discord. we have a bunch of users that maybe can answer your questions.
i would say sure.. if you plugged something while the printer is on it can happen.