ThomasToka/MarlinFirmware #48: Discord

Matisak, 2024-01-06:
Hi, im having some problems! How can I reach you? any discord or only here?

My printer (ender 3 s1 pro) on the v23 does not appear any info almost like you can see and also does not show thumbnails, I have 2(1 32px and other 400px), Not sure what could be the problem.

[image]
[image]

Sorry if I comment on the wrong place im new to git!

ThomasToka, 2024-01-06:
hello,
you need to use my "printdata" postprocessing script.
with those you have nothing will work.
https://github.com/ThomasToka/MarlinFirmware/wiki/Gcode-preview#cura-5x
https://discord.com/invite/Fh4jsUJWe6

closed 2024-01-06 by ThomasToka

Matisak, 2024-01-06:
Thank you so much! I probably missed something! works perfectly!
[image]