ThomasToka/MarlinFirmware #49: Select file is not working

Rex-odus, 2024-01-07:
Hi there,

I flashed your firmware. Everything went well. But now, if I select a file to print, it doesn't show on the home-screen.

What did I do wrong?

Greetings!

Rex

ThomasToka, 2024-01-07:
Hello,
please show me a picture of the about site of your printer.

Rex-odus, 2024-01-07:
Hi Thomas!

Thanks reply!

I can't let you see a picture because I'm back to the stock firmware.

But out of my head it said 023. Your latest. And the screen-version is ending with 105. I can't get the screen flashed. It keeps idle with a line like CRC32 blah. And that's te place I resetted the whole thing after one hour of waiting. Feeling lucky it became no brick. Followed several tutors how to upgrade the firmware. None of them work. So I flashed the whole thing several times. Every time with an inch sweat on my forehead.

Anyway, if you can get me back on track with a working tutor, I'll be in heaven!

Summary:
Your firmware seems to be fine. But as far as I can untangle the story, I need to flash my screen to. And that's not working. I've tried with several combinations of files and directories to the SD-card. None of them work and it always ends with a blue screen with white font and something like CRC32 ...

Just to be sure: Creality Ender 3 S1 Pro is my printer. I've opened it and the chip says STM32F401 RCT6 of RCTb at the end. Not sure with the fonttype used on the chip.

I'm sorry to waste your time. Must be something stupid I do wrong/don't see/do not understand.

Grtz, Rex

[image]

Sent with [Proton Mail](https://proton.me/) secure email.

ThomasToka, 2024-01-07:
```
And the screen-version is ending with 105
```
you need to install naturally my screen also. otherwise it will not work.

ThomasToka, 2024-01-07:
please join the discord and post in #installation-help

closed 2024-01-07 by ThomasToka

Rex-odus, 2024-01-09:
Yeah. Thanks for the link.

Grtz, Patrick

[image]

Sent with [Proton Mail](https://proton.me/) secure email.

Rex-odus, 2024-01-09:
Hi Tomas.

It just hangs with the CRC32 message and I didn't realize it was working already. That's it obviously.

God I hate flashing firmware. Bricked too many things in my life...

But, everything works fine now. Thanks for the nice work!

Let's test...

Grtz, Rex

[image]

Sent with [Proton Mail](https://proton.me/) secure email.