ThomasToka/MarlinFirmware #51: leveling bug

taintremover, 2024-01-11:
I have changed the location of the probe on my s1 pro to the other side of the extruder. I fixed it in the settings menu, in probe offset. Now when I go to level the bed, it will not let me do cr touch probing in manual. It will do a corner. Then will will show that is at 0. The center is the only one that will show a reading. If I do auto leveling, it will show the readings as it doses it. Once it is done all of the readings disappear. I'm on e3s1pro-ubl-v024. I have included a video documenting my problem.  Please let me know if there is any other info you need or how ever I may help you. 

p.s. Im sorry that the video is rotated. When I compressed the video to make it fit as an attachment, it rotated it.

ThomasToka, 2024-01-11:
i dont see a video.
please give me the following values you have set:
probe offsets x an y
home offsets x an y

taintremover, 2024-01-11:
X is 32.5
Y is 0.20

ThomasToka, 2024-01-11:
I can confirm that it does not work with a positive probe offset atm.

I dont know if i manage this to be working with the next release.

So for now its unsupported.

Revert to a firmware that works for now.
Or change your 0 y offset to one on the left side.

taintremover, 2024-01-11:
I just want to say thanks for all your hard work. I really do love the firmware you put out. Keep up the great work!!!

ThomasToka, 2024-01-12:
Hello are you on my discord? If not please join and post in #support.. i have a testbuild for you.

taintremover, 2024-01-12:
what is your discord channel?

ThomasToka, 2024-01-12:
https://discord.com/invite/Fh4jsUJWe6

closed 2024-01-22 by ThomasToka