ThomasToka/MarlinFirmware #52: Script issue with E3S1PROFORKBYTT_printdata_crealityslicer_thumbnail.py

Pweber413, 2024-01-13:
E3S1PROFORKBYTT Thumbnail does not show up after placing the file into the postprocessing plugin script folder. I have restarted 
 Creality slicer 4.8.2 but that script does not appear in the list. 

I did try the file E3S1PROFORKBYTT_Thumbnail.py and it did show up in the list and functioned well.

Any suggestions?

ThomasToka, 2024-01-13:
C:\Program Files\Creality Slicer 4.8.2\plugins\PostProcessingPlugin\scripts

this is the right folder for the script. you can only have one of them in your folder. check if you have the old script somewhere else maybe in AppData/Roaming.

Pweber413, 2024-01-13:
Thank you for the quick response, I'll check for another copy.

Have a great weekend

closed 2024-01-14 by ThomasToka