ThomasToka/MarlinFirmware #55: Print doesn't "finish"

lin-ycv, 2024-01-25:
On v25, when a print is finished, the still shows the pause and stop buttons instead of finish, causing each print to be logged as a cancelled print.

lin-ycv, 2024-01-25:
Also it's possible to set 1 as y margin, which causes crashes.

ThomasToka, 2024-01-25:
The first i could reproduce. Will have a look later where the breaking change was.

But the second: You probably had a display glitch. Then the display beeps takes the value, shows it, but does not transmit it to the display. Cause if it transmit it to the display the mainboard firmware bounced when entering a value lower 10 the min possible value of 10 back. The second part did not happen here. so you have seen the 1 on the display but the margin in the mainboard firmware wasnt changed. Here i can do nothing as in this moment nothing is transferred to the mainboard so i cant make any actions. Please reenter 1 -> it should be bounced to 10. If not do it again.

lin-ycv, 2024-01-25:
For setting the margin, I typed in 0, and it's would show up as 1, after that I ran a bed level, and it crashed.
I'm using the zero y offset mod.

ThomasToka, 2024-01-25:
Show me a pic of your:

probe offsets site 
home offsets site
leveling site with the margins

ThomasToka, 2024-01-25:
I have just uploaded the hotfixed binarys to address the finishing site bug.

The second i could not reproduce. So i need more data like requested earlier today.

closed 2024-01-26 by ThomasToka

lin-ycv, 2024-01-26:
Flashed back to v24 yesterday and did a printed, it just finished. I've now flashed back to v25 using the updated file, and the second issue of safety margin still exist.
[image]
[image]
[image]
In this video you see how close it is to the edge, my home offset is off, so when levelling the rear points it probes outside the bed
https://github.com/ThomasToka/MarlinFirmware/assets/60565737/db0fceff-be4a-4f2a-b4c1-294837ae6a27

ThomasToka, 2024-01-26:
please show me additionally the about site of your printer.

lin-ycv, 2024-01-26:
[image]

ThomasToka, 2024-01-26:
cant reproduce this. :(
thats what i do:

- load firmware
- set home offsets to -2 and -2 over the touch
- set probe offsets to x to -46 and y 0 over the touch
- home printer
- set probes to 1 over the touch
- set x margin to 40, set you margin to 1 -> it reaccounts to 10 over the touch
- start autoleveling

how you do it exactly?

lin-ycv, 2024-01-26:
- Load firmware
- offset home -3,-3
- offset probe -46,0,-1.4
- go to bed level option
- set x margin to 0 (defaults to 40), set y to 0 (defaults to 1)
- start auto level

ThomasToka, 2024-01-26:
ok great now i could reproduce it.

set 1 to y and it will reaccount.

setting 0 does not reaccount in this constellation.

i will fix this also, but dont know if a hotfix or in the next version.

ThomasToka, 2024-01-26:
i have just pushed the hotfix for the 0 y issue:

https://github.com/ThomasToka/MarlinFirmware/releases/tag/MARLIN-E3S1PROFORK-BYTT-v025