ThomasToka/MarlinFirmware #56: error message

UncheckedHook66, 2024-01-25:
I understand this may not be a problem on your end i just don't know what to do as it errors our everytime trying to load firmware onto sd card formatted to specifications on the Readme.txt file. I  was wondering if you may be able to provide insite into this.

ThomasToka, 2024-01-25:
macos user?

UncheckedHook66, 2024-01-25:
sorry im not the best with technical terms, what does that mean

UncheckedHook66, 2024-01-25:
[image]
i have provided a file to show the error message, this happens on copying any of the files

ThomasToka, 2024-01-25:
ok its windows. you have an unpacked folder on your pc and trying to copy the creality stock firmware gives this message?

ThomasToka, 2024-01-25:
please join my discord and screeeaaaam in #installation-help.

we can make a short anydesk session then.

https://discord.com/invite/Fh4jsUJWe6

closed 2024-01-25 by ThomasToka