ThomasToka/MarlinFirmware #60: Screen Update Error

aileach95, 2024-02-11:
Hi! When loading the screen firmware update, everything appeared fine and I got the "END!" message at the end. However, once I booted the printer back up, I ended up with several images that were rendered on top of each other. THen when trying to switch screens, the previous screen information did not dissapear. This resulted in having several layers of screen images from the UI at the same time. I swapped out the microSD card and didn't change anything. So I reverted back to 1.0.5 so my screen would still be usable. 

[image]

ThomasToka, 2024-02-11:
mac user. metadata was written to your sd card and/or your cfg was corrupted.

fix your mac not to write this data to sd or use a windows pc for creating the flash card.

as we have contact on the discord i will close this here.

closed 2024-02-11 by ThomasToka

simonlmn, 2024-06-17:
When trying to install the firmware yesterday I had the exact same problem (note: I was not able to solve the issue and am running the stock firmware again). Tried it with different SD cards and also removing/not creating the "metadata" files but nothing helped. For some reason, when I reverted back to the stock firmware, it did not had any issues with the "metadata" files created by MacOS, so it is a bit strange that it creates such a problem with this firmware. Is there something else that may cause this issue?

As a side note and potential improvement: if the process is so sensitive to the exact formatting and contents of the card, wouldn't it be an option to publish a disk image instead?

ThomasToka, 2024-06-17:
I have heard that so mac installations modify the .cfg file so this becomes not usable.
Some users were able to extract the zip on the sd card directly without moving the files to sd card.
Some other users simply made the card on a windows pc.
As this is a macos problem i wont invest any time in this and not release disk images.
This is a user problem and solvable by the user. Sorry.

dl200010, 2024-06-17:
> mac user. metadata was written to your sd card and/or your cfg was corrupted.
> 
> fix your mac not to write this data to sd or use a windows pc for creating the flash card.
> 
> as we have contact on the discord i will close this here.

The meta data files themselves will not do anything. The printer / screen will ignore them. They are not the exact files in the exact locations they expect, so they are disregarded. Any file that gets modified will be a problem. The .cfg file is probably a text file, and it is possible that MacOS is going into it to make it Mac readable (the whole new line vs carriage return and new line line ending characters is most likely the issue), but doing so makes it unreadable to the printer. Doing a disk image isn't as feasible, because people use the SD card for other things as well and a disk image will need to erase the card to be put on it.

simonlmn, 2024-06-17:
Ok, I get that creating disk images is not high on the priority list. It was just a suggestion as there appear to be also specific requirements on how to format the disk (regarding the block size and so on).

I honestly have never seen MacOS change any file contents just on its own, only when you try to open a file (e.g. in TextEdit or something) the program may or may not convert it (but usually also does not overwrite the file on disk).

To be able to check for myself what the issue is (and hopefully resolve it), could someone give me some kind of hash (MD5 or SHA1) of the config file? I assume it is the file T5LCFG_CRZ2.CFG we are talking about, right?

ThomasToka, 2024-06-17:
the .cfg is a binary and can only be edited with a hex editor. the md5sums are always in the release.