ThomasToka/MarlinFirmware #62: Printing does not start...after Update from Stock (S1 Pro)

njordan77, 2024-02-21:
Hi,
i recently switched from stock S1 Pro image to your FW.
I even opened the case to see the STM32 chip, which is F4 in my case.
Update worked easy. already had Display 1.05 uploaded. I can use the touch screen,
also do the leveling (btw. not sure if its just me, but level point 5 always moves to point 4??)
and the HW seems to react to what i select on the display.

BUT when i choose a previous gcode file from SD it simply does not start when pressing on "PLAY" button. nothing, no reaction, simply does not start.

Any idea what is the reason, I tried several files, several SD cards....
Thanks for your thoughts. I use the most recent v25 file.

Kind Regards,

njordan77, 2024-02-21:
[image]

dl200010, 2024-02-21:
> I know its screen version 1.05 but not your specific firmware for the display. but i tried it also with your display firmware, does not matter.
> 
> ![Untitled](https://private-user-images.githubusercontent.com/7437150/306643294-d5f475b4-cfee-45f5-9891-69a2164bffab.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDg1MjM0MDMsIm5iZiI6MTcwODUyMzEwMywicGF0aCI6Ii83NDM3MTUwLzMwNjY0MzI5NC1kNWY0NzViNC1jZmVlLTQ1ZjUtOTg5MS02OWEyMTY0YmZmYWIuanBnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDIyMSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAyMjFUMTM0NTAzWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZWMyNDMzNTI1MjM4YTg4ZDQwOGIxZTg4ZWYwYjliZWE2YmMyYjkwOWQxZmQ4ZTg4ZWMwOTBjYzQwYjFjNmJjNyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.238jgSYSMMBa3yy4pKbP90NX1IgdZP19peSeQKaWd-w)

You do need to update the screen. The SD card may be rejected for not being the right partition size, format, or just the screen hates the card. Head over to the discord in the instructions.txt and many there will help you fix it.