ThomasToka/MarlinFirmware #68: Print button not working

sudoaptinstall, 2024-03-10:
The print button on the screen does not work. Every single button in the system works except it. I want to print with SD card but I cant because the Print button does not work when clicked. For now, I can only print via USB cable

ThomasToka, 2024-03-10:
You dont have my screen firmware installed.
Have a look at the about site of your printer.
You should see a qrcode and v025 as screen version.
If you need help please come to the discord.

closed 2024-03-10 by ThomasToka

sudoaptinstall, 2024-03-10:
I do have your screen firmware installed, printer says: (Screen VER: v2. 0. 0-v025-byTT).

ThomasToka, 2024-03-10:
Please show a pic of the about site?

sudoaptinstall, 2024-03-10:
[image]

ThomasToka, 2024-03-10:
The something else is fishy. 

Hundreds of users use it. Reset you printer. If this does not helpp please go to my discord. Post this in #printing-help.

We will try to help you.

sudoaptinstall, 2024-03-10:
Where do I find the Discord link?

ThomasToka, 2024-03-10:
installation.txt