ThomasToka/MarlinFirmware #69: Setting the flow during printing

ansei91, 2024-03-15:
Hello, dear!
I have been using your firmware for a long time since version v008.
I set up my Ender 3 S1 pro printer perfectly.
There is a problem. Flow change does not work on any of the firmwares. When I want to change the flow while printing, it simply does not change. But it will change after I start a new print. The percentage that I set in the previous print is used only after the next print is launched. I hope you understand what I'm talking about. Is it possible to make the flow change during the printing process?
Thank you very much for your hard work!

ThomasToka, 2024-03-15:
Sorry if it does not work "live" it wont work and is not injectable into the queue. Its not a firmware bug, but by design this way.
I cant do anything here then atm.

closed 2024-03-15 by ThomasToka

ansei91, 2024-03-15:
It is very inconvenient when this function does not work.
Is it possible to make this function work in real time?

ansei91, 2024-03-15:
Tell me what this icon includes?
[image]

ThomasToka, 2024-03-15:
Please read my last answer regarding the flow again.

Please read the wiki: https://github.com/ThomasToka/MarlinFirmware/wiki/Display-Pages

The issue tracker is not for random questions about the firmware. Please use the discord for your questions.