ThomasToka/MarlinFirmware #75: Problems with z axis after power outage

nikoskiriazidis, 2024-06-21:
Hi, when the power comes back on after an outage, the printer asks me if I want to continue printing. I press continue and then everything seems to be going well (x and y axis homing), but when the printhead moves to the printing part, it smashes the toolhead into the part. Specifically, it seems like the z-axis doesn't know where to resume printing. Any ideas what the problem might be? Thank you.

ThomasToka, 2024-06-21:
Hello,
maybe there have some commits interfering the process.
My time this month is very limited.
You can help doing the following:
- please install v015 as in this release a power loss recovery fix was introduced. you need to install the mainboard and screen for this test. sorry.
- retry the process. does it work like it should?
- if this works please install v024 mainboard and sceen and do the power outage test again. does it still work?
- if v024 still works. install v025.. test.. if this works install v026 and test again.

please tell me then which was the last release that worked for you.

thank your for reporting and helping to break this down.

nikoskiriazidis, 2024-06-21:
thanks, I will try and let you know

ThomasToka, 2024-07-13:
Closed as no further informations were delivered.

closed 2024-07-13 by ThomasToka