ThomasToka/MarlinFirmware #76: SD card crashes printer

lin-ycv, 2024-07-04:
Not sure what the issue is but if printer crashes a lot with v27 when inserting ad card, reading from SD card.
Sometimes having the ad card inserted at boot would cause the printer to hang at boot.

closed 2024-07-04 by ThomasToka

ThomasToka, 2024-07-04:
Now i have a little more time to answer.

Your bug is maybe that what i have seen very selden. In the time i build this firmware such a thing was reported maybe twice or three times of over 4000 user interactions. So the number is low. For those users it went away as quickly as it appeared.

Sometimes the sd card reader refuses to read cards temporary. I dont know if its the connection inside the cardreader to the pins or some sort of card failures.

I know that some few users had this without a hardware failure. those other that had something like this could fix it with new cards or a new mainboard (with new cardreader).

There are over 650 users meanwhile on v027 and you are the first and only to report this for months. I even dont know when the last user reported this. It must have been 8 months or so.

lin-ycv, 2024-07-05:
My SD cards might be wearing out, or maybe I have power issues, not sure. Printers sometimes crash mid printer repeatedly, but other times prints hours without issue. 
Could also be overheating? It's getting hot these days. Not sure, need to observe more to figure out.

But you are right, it's not a firmware issue.