ThomasToka/MarlinFirmware #78: lift on pause?

lin-ycv, 2024-07-12:
I might be wrong, but I remember pause used to instruct the printer to lift the nozle slightly then goto the parked position.
On v27, pause move horizontally to the parked position without lifting.

If lifting on pause was never a feature, is it possible for it to be added?

dl200010, 2024-07-12:
Have you tried to turn on Z hop on retract? I do this and I still get the pause to lift that amount and then park.

lin-ycv, 2024-07-12:
it's on auto in orcaslicer, i'll test other explicit options next week and see if that makes a difference.
[image]

dl200010, 2024-07-12:
I only ask because I have not seen this. I have Z hop on always.

ThomasToka, 2024-07-13:
This function relays on marlin functions.

If you a see a change please tell me when exactly this behavior changed. I need the exact change version like "v014 to v015".

I will then have a look in it.

Sorry i have no time for such minor things and can not invest any time to localise the change.

When i know where it changed i maybe will have a look in it and reverse it. But i will then tell you if i had a look in the change.

closed 2024-07-13 by ThomasToka