ThomasToka/MarlinFirmware #82: Termination of printing when the ambient temperature exceeds 50 degrees

kmaras77, 2024-10-02:
The printer Creality Ender 3 S1 Pro, FW v027 is used in an enclosure so that the prints do not distort. I have a temperature measurement in this enclosure and if the temperature exceeds 50 degrees, the printer stops working and the print shows 100% and ends. Is there an option to raise this temperature to 60 degrees or add such a configuration option to work in the enclosure?

ThomasToka, 2024-10-02:
I think this is not a firmware thing.

Additionally this is not a user error or hardware error issue tracker.

Please use our discord, maybe one of our members has an idea.

closed 2024-10-02 by ThomasToka

kmaras77, 2024-10-03:
This is related to these variables and code fragments:

AUTO_POWER_CONTROL

AUTO_POWER_E_TEMP

#if ANY(AUTO_POWER_CONTROL, POWER_OFF_WAIT_FOR_COOLDOWN)
    //#define AUTO_POWER_E_TEMP        50 // (°C) PSU on if any extruder is over this temperature
    //#define AUTO_POWER_CHAMBER_TEMP  30 // (°C) PSU on if the chamber is over this temperature
    //#define AUTO_POWER_COOLER_TEMP   26 // (°C) PSU on if the cooler is over this temperature
  #endif