ThomasToka/MarlinFirmware #84: Support for Ender 3 S1 ProX

Waldehyd, 2024-10-20:
I got a new 2024 Version of the Ender 3 S1 Pro called ProX. It supposedly prints faster, but only due to Firmware changes. Do you have any information on whether it's a good idea to flash it with a new Firmware?

ThomasToka, 2024-10-20:
Any links to the machine or its used firmware.

Can you connect with a terminal and show me the output of:

M503
M115

ThomasToka, 2024-10-20:
It seems its only a marketing thing to call it ProX with the F28 firmware.

https://forum.creality.com/t/difference-between-ender-3-s1-pro-and-ender-3-s1-prox/8957/9

So its supported out of the box with my firmware.

If you can provide the M503 and M115 for inspection.

For now i will close this here. You can still answer here.

closed 2024-10-20 by ThomasToka

Waldehyd, 2024-10-20:
I see, thtanks, for completeness:

Send: M503
Recv: echo:  G21    ; Units in mm (mm)
Recv: 
Recv: echo:; Filament settings: Disabled
Recv: echo:  M200 S0 D1.75
Recv: echo:; Steps per unit:
Recv: echo: M92 X80.00 Y80.00 Z400.00 E424.90
Recv: echo:; Maximum feedrates (units/s):
Recv: echo:  M203 X500.00 Y500.00 Z10.00 E25.00
Recv: echo:; Maximum Acceleration (units/s2):
Recv: echo:  M201 X5000.00 Y5000.00 Z100.00 E5000.00
Recv: echo:; Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
Recv: echo:  M204 P500.00 R1500.00 T500.00
Recv: echo:; Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>
Recv: echo:  M205 B20000.00 S0.00 T0.00 X15.00 Y15.00 Z0.40 E5.00
Recv: echo:; Home offset:
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:; Auto Bed Leveling:
Recv: echo:  M420 S1 Z10.00
Recv: echo:  G29 W I0 J0 Z0.18500
Recv: echo:  G29 W I1 J0 Z0.25650
Recv: echo:  G29 W I2 J0 Z0.17400
Recv: echo:  G29 W I3 J0 Z-0.09950
Recv: echo:  G29 W I0 J1 Z0.15400
Recv: echo:  G29 W I1 J1 Z0.12700
Recv: echo:  G29 W I2 J1 Z0.04500
Recv: echo:  G29 W I3 J1 Z-0.14200
Recv: echo:  G29 W I0 J2 Z0.09400
Recv: echo:  G29 W I1 J2 Z0.06150
Recv: echo:  G29 W I2 J2 Z-0.01400
Recv: echo:  G29 W I3 J2 Z-0.19600
Recv: echo:  G29 W I0 J3 Z0.06600
Recv: echo:  G29 W I1 J3 Z0.13600
Recv: echo:  G29 W I2 J3 Z0.06900
Recv: echo:  G29 W I3 J3 Z-0.14900
Recv: echo:; Material heatup parameters:
Recv: echo:  M145 S0 H200.00 B60.00 F255
Recv: echo:  M145 S1 H240.00 B110.00 F255
Recv: echo:; PID settings:
Recv: echo:  M301 P30.82 I4.19 D56.70
Recv: echo:  M304 P174.91 I34.70 D587.69
Recv: echo:; Power-Loss Recovery:
Recv: echo:  M413 S1
Recv: echo:; Z-Probe Offset (mm):
Recv: echo:  M851 X-31.80 Y-40.50 Z-2.75
Recv: Input Shaping:
Recv:   M593 X F50.00 D0.15
Recv:   M593 Y F35.00 D0.15
Recv: echo:; Filament load/unload lengths:
Recv: echo:  M603 L0.00 U100.00
Recv: echo:; Filament runout sensor:
Recv: echo:  M412 S1
Recv: echo:; UI Language:
Recv: echo:  M414 S0
______
Recv: FIRMWARE_NAME:Marlin 2.0.8.28F4 (Jun 19 2023 18:06:53) SOURCE_CODE_URL:github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:Ender-3 S1 Pro EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
Recv: Cap:SERIAL_XON_XOFF:0
Recv: Cap:BINARY_FILE_TRANSFER:0
Recv: Cap:IS_PLR:0
Recv: Cap:EEPROM:1
Recv: Cap:VOLUMETRIC:1
Recv: Cap:AUTOREPORT_POS:0
Recv: Cap:AUTOREPORT_TEMP:1
Recv: Cap:PROGRESS:0
Recv: Cap:PRINT_JOB:1
Recv: Cap:AUTOLEVEL:1
Recv: Cap:RUNOUT:1
Recv: Cap:Z_PROBE:1
Recv: Cap:LEVELING_DATA:1
Recv: Cap:BUILD_PERCENT:0
Recv: Cap:SOFTWARE_POWER:0
Recv: Cap:TOGGLE_LIGHTS:0
Recv: Cap:CASE_LIGHT_BRIGHTNESS:0
Recv: Cap:EMERGENCY_PARSER:0
Recv: Cap:HOST_ACTION_COMMANDS:0
Recv: Cap:PROMPT_SUPPORT:0
Recv: Cap:SDCARD:1
Recv: Cap:REPEAT:0
Recv: Cap:SD_WRITE:1
Recv: Cap:AUTOREPORT_SD_STATUS:0
Recv: Cap:LONG_FILENAME:1
Recv: Cap:THERMAL_PROTECTION:1
Recv: Cap:MOTION_MODES:0
Recv: Cap:ARCS:1
Recv: Cap:BABYSTEPPING:1
Recv: Cap:CHAMBER_TEMPERATURE:0
Recv: Cap:COOLER_TEMPERATURE:0
Recv: Cap:MEATPACK:0
Recv: ok

ThomasToka, 2024-10-20:
As expected. Same firmware as on all other pros.

Thanks for the outputs.