ThomasToka/MarlinFirmware #85: V29 ABL Z Offset not saving

lin-ycv, 2024-11-01:
As seen in the video, once the Z height is set, and i shut down the printer, it resets back to 0
https://github.com/user-attachments/assets/71199940-4874-4c0a-a2ab-b4b38565b6e9

ThomasToka, 2024-11-01:
Hello,
this change was done in v027 to save saves on the eeprom and is for now intended.
The save to the eeprom is done after the autoleveling run.
I will think about it if i change this to another behavior. 
But everything i do will penetrate the eeprom unneeded again.

So for now: Not a bug but a feature.

Stay tuned for v030. Maybe i touch this again.

closed 2024-11-01 by ThomasToka

lin-ycv, 2024-11-01:
I see, maybe the easiest solution is to add a button for saving to eeprom?