ThomasToka/MarlinFirmware #86: ENDER 3 S1 Plus  bed dimension ?

jujumo, 2025-01-31:
Hi,  Thanks for this awesome work.

I just bought a second hand S1 Plus, and installed  "firmware-20241227-193454-E3S1Plus-ubl-v031-F4.bin"
I noticed that the Y axis hit the end of range while doing bed leveling.
In the config menu, I noticed print size is "310 * 310 * 300".
But on official specs (https://www.creality.com/products/creality-ender-3-s1-plus-3d-printer)
bed size is "300 * 300 * 300". 

Is that expected ?

dl200010, 2025-01-31:
Yes. The stock size has a buffer around the bed. This firmware opens up so you can print edge to edge.

closed 2025-01-31 by dl200010

jujumo, 2025-01-31:
Thanks for quick reply. So, any clue of what could cause the out of range hit while leveling ?

ThomasToka, 2025-01-31:
You need to read the Installation.txt. Especially the Part "After the Installation". I case of user error questions like this (you did not read the provided instructions, so you did not calibrate your home offsets and/or purge lines or did it wrong) please use one of the the discord channels like #installation-help or #printing-basics. The link to the discord is also in the 

This issue tracker here is not for calibration or configuration questions!

jujumo, 2025-01-31:
Off course, I quit reading too soon. Thanks for pointing this.

dl200010, 2025-02-01:
> Off course, I quit reading too soon. Thanks for pointing this.

You should join the discord. There are many on there that will help you. I am one of them.